Part-DB/Part-DB-server #205: Example nginx configuration?

suuppl, 2023-01-27:
**Is your feature request related to a problem? Please describe.**
I struggle to setup part-db using nginx.

**Describe the solution you'd like**
I would like an example configuration file for Part-DB on nginx. I think the best spot would be the Ubuntu configuration manual.
(preferably with ssl support, though i will manage without)

**Describe alternatives you've considered**
I've thought about serving it with an apache server and only proxying via nginx, but then I'd have apache and nginx running.
(and my server is quite limited, so I would like to avoid that)

Thanks for any help :)

suuppl, 2023-01-30:
After installing php8.1-fpm (`sudo apt install php8.1-fpm`) I got it to work with this configuration:
```
server {
    listen 80;
    server_name parts.example.com;
    return 301 https://$host$request_uri;
}
server {
#   listen 80;
    listen 443 ssl;

    server_name parts.example.com;
    root /var/www/partdb/public;

    location / {
        try_files $uri /index.php$is_args$args;
    }

    location ~ ^/index\.php(/|$) {
        fastcgi_pass unix:/var/run/php/php-fpm.sock;
        fastcgi_split_path_info ^(.+\.php)(/.+)$;
        include fastcgi_params;
        fastcgi_param SCRIPT_FILENAME $realpath_root$fastcgi_script_name;
        fastcgi_param DOCUMENT_ROOT $realpath_root;

        internal;
    }

    location ~ \.php$ {
        return 404;
    }

    error_log /var/log/nginx/parts.error.log;
    access_log /var/log/nginx/parts.access.log;

    # SSL parameters
    ssl_certificate /var/www/certs/SSL/domain.cert.pem;
    ssl_certificate_key /var/www/certs/SSL/private.key.pem;
    ssl_trusted_certificate /var/www/certs/SSL/intermediate.cert.pem;

    ssl_session_timeout 5m;

    ssl_protocols TLSv1.2;
    ssl_ciphers HIGH:!aNULL:!MD5;
    ssl_prefer_server_ciphers on;
}
```

If there are no issues with this configuration, you can use it as an example, if you want.

I will update this comment if I find any problems with that config.

closed 2023-01-30 by suuppl

jbtronics, 2023-01-30:
Looks fine, it is very similar to the minimal configutation example of Symfony.
https://symfony.com/doc/current/setup/web_server_configuration.html

Does the url rewrite works properly with that config? So can you call `https://parts.example.com/en/part/new/` or do you need to call `https://parts.example.com/index.php/en/part/new/

suuppl, 2023-01-30:
It works fine. The url with `index.php` 404s.