Part-DB/Part-DB-server #206: Docker image is not working

Ziesie1, 2023-01-30:
**Describe the bug**
After going through the steps described in the [Docker description](https://github.com/Part-DB/Part-DB-symfony/blob/master/docs/docker/docker-install.md), I encounter an error in step 4 that the database cannot be opened.

**To Reproduce**
Steps are described in the [Docker description](https://github.com/Part-DB/Part-DB-symfony/blob/master/docs/docker/docker-install.md)

**Output**

```
$ docker exec --user=www-data partdb php bin/console doctrine:migrations:migrate

In ExceptionConverter.php line 76:

  An exception occurred in the driver: SQLSTATE[HY000] [14] unable to open da
  tabase file

In Exception.php line 28:

  SQLSTATE[HY000] [14] unable to open database file

In Driver.php line 39:

  SQLSTATE[HY000] [14] unable to open database file

2023-01-30T08:30:22+00:00 [info] User Deprecated: Method "Symfony\Component\HttpKernel\Bundle\Bundle::getContainerExtension()" might add "?ExtensionInterface" as a native return type declaration in the future. Do the same in child class "Omines\DataTablesBundle\DataTablesBundle" now to avoid errors or add an explicit @return annotation to suppress this message.
2023-01-30T08:30:22+00:00 [info] User Deprecated: Method "Symfony\Component\HttpKernel\Bundle\Bundle::getContainerExtension()" might add "?ExtensionInterface" as a native return type declaration in the future. Do the same in child class "SpomkyLabs\CborBundle\SpomkyLabsCborBundle" now to avoid errors or add an explicit @return annotation to suppress this message.
2023-01-30T08:30:22+00:00 [info] User Deprecated: Method "Symfony\Component\HttpKernel\Bundle\Bundle::getContainerExtension()" might add "?ExtensionInterface" as a native return type declaration in the future. Do the same in child class "Webauthn\Bundle\WebauthnBundle" now to avoid errors or add an explicit @return annotation to suppress this message.
2023-01-30T08:30:22+00:00 [info] User Deprecated: The Liip\ImagineBundle\Templating\FilterExtension class is deprecated since version 2.7 and will be removed in 3.0; configure "liip_imagine.twig.mode" to "lazy" instead.
2023-01-30T08:30:22+00:00 [info] User Deprecated: The Liip\ImagineBundle\Templating\FilterTrait trait is deprecated since version 2.7 and will be removed in 3.0; use Twig instead.
2023-01-30T08:30:22+00:00 [info] User Deprecated: Method "Twig\Extension\ExtensionInterface::getFilters()" might add "array" as a native return type declaration in the future. Do the same in implementation "Liip\ImagineBundle\Templating\FilterExtension" now to avoid errors or add an explicit @return annotation to suppress this message.
2023-01-30T08:30:22+00:00 [info] User Deprecated: Since symfony/security-core 5.3: The "Symfony\Component\Security\Core\Authentication\Provider\AuthenticationProviderInterface" interface is deprecated, use the new authenticator system instead.
2023-01-30T08:30:22+00:00 [info] User Deprecated: The "Webauthn\Bundle\Security\Authentication\Provider\WebauthnProvider" class implements "Symfony\Component\Security\Core\Authentication\Provider\AuthenticationProviderInterface" that is deprecated since Symfony 5.3, use the new authenticator system instead.
```

jbtronics, 2023-01-30:
Are the folders you are mapping existing on the host system? (Should be in the folder in which you run `docker-compose up`)

Ziesie1, 2023-01-30:
Yes, the folders have been created but are empty.

Ziesie1, 2023-01-30:
I was able to reproduce the problem on two different Linux (Ubuntu 22.04) machines. Same behavior.

jbtronics, 2023-02-03:
This is caused by wrong permissions inside the docker container. The volume folders are created with root permission, and the apache web user, can not write them.

As a workaround, you can explicitly chown the folders in the container:

`docker exec partdb chown www-data:www-data /var/www/html/uploads`
`docker exec partdb chown www-data:www-data /var/www/html/var/db`
and `docker exec partdb chown www-data:www-data /var/www/html/public/media`

I don't have much experience with docker, but  I will try to find a better fix (so that these steps are not needed)

jbtronics, 2023-02-04:
The new docker images should now work out of the box.

closed 2023-02-04 by jbtronics