Part-DB/Part-DB-server #217: Marking ordered components "waiting to arrive"

wu5, 2023-02-16:
First of all, I want to thank you for the good work. I really like the direction this project is taking.

Right now I've created an inventory "Ordered", but this is not very convenient, because these parts are added to the total number of parts available. It would also be useful to be able to show the entire list of components that are already being delivered. 

In addition, I would like to see more information about the parts amounts on the project's build page under the "Build" tab. To more clearly understand how much I have in stock and how much I need to order.

If you paid attention to these aspects, it would be just wonderful!

jbtronics, 2023-02-16:
I am planning to implement some proper ordering system, were you can mark parts for ordering, create order data objects, with information about what was ordered when (with metadata like delivery status, order number and so on), and the possibility to automatically add the ordered parts to your stock in the end. However that might take some time to implement and develop.

As a workaround you could maybe use tags, and add tags like "need to be ordered", "in delivery", etc. to your parts.

To the build information: If you want to know how many parts are missing to build a certain amount of a project, just type in your wished build quantity and try to build it. The build page will then show you exactly what amounts are missing.

amitzioni, 2024-01-17:
I want also first thank you for the good work and fantastic documentation. 
Is there a prospective date for the  implementation of the ordering system ?

amitzioni, 2024-01-23:
> I am planning to implement some proper ordering system, were you can mark parts for ordering, create order data objects, with information about what was ordered when (with metadata like delivery [...]

I want also first thank you for the good work and fantastic documentation.
Is there a prospective date for the implementation of the ordering system ?

jbtronics, 2024-01-23:
> Is there a prospective date for the implementation of the ordering system ?

Not really. Currently I am working on some system to change settings/configure Part-DB from the WebUI instead of requiring env variables for everything.

Maybe I will look into the ordering system after that.