Part-DB/Part-DB-server #218: Error 500 when filtering by stock level with demo version

StormTrooper, 2023-02-18:
**Describe the bug**
Error 500 when trying to filter by stock level

**To Reproduce**
Steps to reproduce the behavior:

This error is occuring on the http://part-db.herokuapp.com/
I am demo'ing this and want to be sure it does what I want  before I spend the time installing on physical hardware and then run into this issue. So not sure if this will also occur on a physical install. 

1. Click on Tools / Show / All parts. 
2. Click Filter / Stocks
3. Set total amount < 5
4. Click Update
5. Internal Server Error (Status 500)

```
500
Internal Server Error
Something bad happened internally.
There is nothing you could do about this, except trying to reload the page.
You can try to [Go Back](javascript:history.back()) or [Visit the homepage](http://part-db.herokuapp.com/en/).

If this error persists, please contact your administrator.

Infos for Admin: Invalid or not existing database schema.
Try following things:

Check if the DATABASE_URL in .env.local is correct
Run php bin/console doctrine:migrations:migrate to upgrade database schema
Run php bin/console cache:clear
If you think that this is an error in Part-DB, open an Issue on [GitHub](https://github.com/Part-DB/Part-DB-symfony/issues).

You can disable these hints in config/parameters.yaml by setting partdb.error_pages.show_help to false.

This page was generated by [Part-DB](https://github.com/Part-DB/Part-DB-symfony) on Feb 18, 2023, 10:15 PM .
````

**Expected behavior**
Expect filter to show all items < 5 

**Screenshots**

[image]

**Server Side**
http://part-db.herokuapp.com/

jbtronics, 2023-02-19:
This seems to be the detailed error message from the log: 

```
Feb 18 22:07:30 part-db app/web.1: {"message":"Uncaught PHP Exception Doctrine\\DBAL\\Exception\\InvalidFieldNameException: \"An exception occurred while executing a query: SQLSTATE[42S22]: Column not found: 1054 Unknown column 'sclr_0' in 'where clause'\" at /app/vendor/doctrine/dbal/src/Driver/API/MySQL/ExceptionConverter.php line 67","context":{"exception":{"class":"Doctrine\\DBAL\\Exception\\InvalidFieldNameException","message":"An exception occurred while executing a query: SQLSTATE[42S22]: Column not found: 1054 Unknown column 'sclr_0' in 'where clause'","code":1054,"file":"/app/vendor/doctrine/dbal/src/Driver/API/MySQL/ExceptionConverter.php:67","trace":["/app/vendor/doctrine/dbal/src/Connection.php:1931","/app/vendor/doctrine/dbal/src/Connection.php:1874","/app/vendor/doctrine/dbal/src/Connection.php:1099","/app/vendor/doctrine/orm/lib/Doctrine/ORM/Query/Exec/SingleSelectExecutor.php:31","/app/vendor/doctrine/orm/lib/Doctrine/ORM/Query.php:325","/app/vendor/doctrine/orm/lib/Doctrine/ORM/AbstractQuery.php:1212","/app/vendor/doctrine/orm/lib/Doctrine/ORM/AbstractQuery.php:1166","/app/vendor/doctrine/orm/lib/Doctrine/ORM/AbstractQuery.php:937","/app/vendor/doctrine/orm/lib/Doctrine/ORM/Tools/Pagination/Paginator.php:117","/app/vendor/omines/datatables-bundle/src/Adapter/Doctrine/FetchJoinORMAdapter.php:126","/app/vendor/omines/datatables-bundle/src/Adapter/Doctrine/FetchJoinORMAdapter.php:83","/app/vendor/omines/datatables-bundle/src/Adapter/AbstractAdapter.php:45","/app/vendor/omines/datatables-bundle/src/DataTable.php:338","/app/vendor/omines/datatables-bundle/src/DataTable.php:300","/app/src/Controller/PartListsController.php:147","/app/src/Controller/PartListsController.php:337","/app/vendor/symfony/http-kernel/HttpKernel.php:163","/app/vendor/symfony/http-kernel/HttpKernel.php:75","/app/vendor/symfony/http-kernel/Kernel.php:202","/app/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php:35","/app/vendor/autoload_runtime.php:35","/app/public/index.php:5"],"previous":{"class":"Doctrine\\DBAL\\Driver\\PDO\\Exception","message":"SQLSTATE[42S22]: Column not found: 1054 Unknown column 'sclr_0' in 'where clause'","code":1054,"file":"/app/vendor/doctrine/dbal/src/Driver/PDO/Exception.php:28","trace":["/app/vendor/doctrine/dbal/src/Driver/PDO/Statement.php:123","/app/vendor/doctrine/dbal/src/Connection.php:1092","/app/vendor/doctrine/orm/lib/Doctrine/ORM/Query/Exec/SingleSelectExecutor.php:31","/app/vendor/doctrine/orm/lib/Doctrine/ORM/Query.php:325","/app/vendor/doctrine/orm/lib/Doctrine/ORM/AbstractQuery.php:1212","/app/vendor/doctrine/orm/lib/Doctrine/ORM/AbstractQuery.php:1166","/app/vendor/doctrine/orm/lib/Doctrine/ORM/AbstractQuery.php:937","/app/vendor/doctrine/orm/lib/Doctrine/ORM/Tools/Pagination/Paginator.php:117","/app/vendor/omines/datatables-bundle/src/Adapter/Doctrine/FetchJoinORMAdapter.php:126","/app/vendor/omines/datatables-bundle/src/Adapter/Doctrine/FetchJoinORMAdapter.php:83","/app/vendor/omines/datatables-bundle/src/Adapter/AbstractAdapter.php:45","/app/vendor/omines/datatables-bundle/src/DataTable.php:338","/app/vendor/omines/datatables-bundle/src/DataTable.php:300","/app/src/Controller/PartListsController.php:147","/app/src/Controller/PartListsController.php:337","/app/vendor/symfony/http-kernel/HttpKernel.php:163","/app/vendor/symfony/http-kernel/HttpKernel.php:75","/app/vendor/symfony/http-kernel/Kernel.php:202","/app/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php:35","/app/vendor/autoload_runtime.php:35","/app/public/index.php:5"],"previous":{"class":"PDOException","message":"SQLSTATE[42S22]: Column not found: 1054 Unknown column 'sclr_0' in 'where clause'","code":42,"file":"/app/vendor/doctrine/dbal/src/Driver/PDO/Statement.php:121","trace":["/app/vendor/doctrine/dbal/src/Driver/PDO/Statement.php:121","/app/vendor/doctrine/dbal/src/Connection.php:1092","/app/vendor/doctrine/orm/lib/Doctrine/ORM/Query/Exec/SingleSelectExecutor.php:31","/app/vendor/doctrine/orm/lib/Doctrine/ORM/Query.php:325","/app/vendor/doctrine/orm/lib/Doctrine/ORM/AbstractQuery.php:1212","/app/vendor/doctrine/orm/lib/Doctrine/ORM/AbstractQuery.php:1166","/app/vendor/doctrine/orm/lib/Doctrine/ORM/AbstractQuery.php:937","/app/vendor/doctrine/orm/lib/Doctrine/ORM/Tools/Pagination/Paginator.php:117","/app/vendor/omines/datatables-bundle/src/Adapter/Doctrine/FetchJoinORMAdapter.php:126","/app/vendor/omines/datatables-bundle/src/Adapter/Doctrine/FetchJoinORMAdapter.php:83","/app/vendor/omines/datatables-bundle/src/Adapter/AbstractAdapter.php:45","/app/vendor/omines/datatables-bundle/src/DataTable.php:338","/app/vendor/omines/datatables-bundle/src/DataTable.php:300","/app/src/Controller/PartListsController.php:147","/app/src/Controller/PartListsController.php:337","/app/vendor/symfony/http-kernel/HttpKernel.php:163","/app/vendor/symfony/http-kernel/HttpKernel.php:75","/app/vendor/symfony/http-kernel/Kernel.php:202","/app/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php:35","/app/vendor/autoload_runtime.php:35","/app/public/index.php:5"]}}}},"level":500,"level_name":"CRITICAL","channel":"request","datetime":"2023-02-18T21:07:29.681847+00:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"user","username":"user"},"url":"/en/parts?part_filter%5BamountSum%5D%5Boperator%5D=%3C&part_filter%5BamountSum%5D%5Bvalue1%5D=5&part_filter%5Bsubmit%5D=","ip":"10.1.43.219","http_method":"POST","server":"part-db.herokuapp.com","referrer":"http://part-db.herokuapp.com/en/parts?part_filter%5BamountSum%5D%5Boperator%5D=%3C&part_filter%5BamountSum%5D%5Bvalue1%5D=5&part_filter%5Bsubmit%5D="}}
```

jbtronics, 2023-02-20:
The virtual field which holds the amountSum of a part contains an aggregate and therefore cannot be used in a WHERE clause (like it was used before). The constraint was moved into the HAVING clause of the query.

The issue is fixed in Release 1.0.1

closed 2023-02-20 by jbtronics