Part-DB/Part-DB-server #221: Handle owner of a part

mase76, 2023-02-23:
Hi,
for now I handle the owner of a part with a note. It would be nice, if there were an owner management. This makes sense when using the system in a hackerspace, where people store their private stuff and people can make requests for parts.

jbtronics, 2023-02-24:
In principle it should not be difficult to add a new field ("owner" or something like this) to parts.

However there are some design choices to be made: Can a part only have a single owner or multiple ones? Must the owner be one of the registered users, or can it be basically any string (which would not be a big improvement over just putting it in the notes section, semantically)?

Also can the different stocks (on different locations) of a part can have different owners?

What is your opinion on this?

mase76, 2023-02-24:
It depends on how far you want to go. In our hackerspace we have some personal storage places. All part stocks, that go in there, could belong to the storage owner automatically. However a per stock ownership should be enough. I can have 10 arduinos in a stock, that belongs to me, and further arduinos in an other stock, that is public. Maybe only the owner should be able to do a  retrieval, except the stock is public. I think, multiple owners are not necessary. And I think, reducing it to the registered users is enough. Maybe some more contact data like chat address in addition to the email.

jbtronics, 2023-04-04:
> I have an issue when adding the owner to a label profile. I get an server error 500 after that. The label works, but I cannot edit it any more, because I always run in that server error.

This should be fixed now.

> Also the owner is missing in the label profile for storage places.

Also implemented.

closed 2023-04-04 by mase76