Part-DB/Part-DB-server #222: Use footprint names from Kicad

mase76, 2023-02-25:
Hi,
I recommend to use the footprint names from Kicad. As Kicad is the most used PCB editor, it would make the part finding easier. If you agree, I would help with the renaming.

jbtronics, 2023-02-25:
What exactly do you wanna rename? The footprints as datastructure are user defined and you can name them however you want.

Do you mean the builtin footprint images? I'm not quite sure if that would be useful, as they maybe not all match KiCADs footprints and maybe have some subtile differences.
As far as I know (these images were already part of the older versions of Part-DB before I took over the project) the footprints images were generated automatically from the models of Eagle3D, which offered 3D models for various Eagle footprints.

In the legacy version there were the feature to assign 3D models to footprint datastructures (and view the 3D models directly in Part-DB). The shipped footprints were basically just converted versions of the KiCAD 3D library.
This feature is not included in the new versions yet, as I deemed it not very important, and I want to use another viewer library which supports more than the rather obscure x3d format as in the legacy Part-DB version.

mase76, 2023-02-25:
Yes, I mean the builtin footprint images. For example the filenames do not contain data about the body size. I cannot distinguish SOIC-8_3.9x4.9mm_P1.27mm and SOIC-8_5.3x5.3mm_P1.27mm. Of course I can name them freely, but  it would be easier to find them. I don't know, how they are named in Eagle.