Part-DB/Part-DB-server #226: separate permission for export function

sascha988, 2023-02-25:
A separate user permission for the export function would be great.
Right now it seems that if you set the permission to "view", you also give the ability to export.

Tested with URL: https://URL-TO-PATH-DB/de/category/export

jbtronics, 2023-02-25:
Indeed the export ability is currently coupled to the view permission. I don't really understand the need to split these up, as the export should not contain any more info than the user should be able to see from the web interface.
So basically even if the user would not have this export permission he would be able to create a file with the same content as generated with the export function by clicking through every data mode and copy the data (or writing a script which scrapes the data automatically).
So there is no really benefit from preventing exports for users with view permission from a security point of view.

sascha988, 2023-02-26:
Hello Jan,
I was thinking about this again, and of course, you're right... My thoughts about that was to make it harder to gather a lot of Information in a easy way, but yes, if someone really wants them, they can get them... So no need for extra work :)
Thanks a lot for your thoughts on that and for your fantastic part-db project :)
Sascha

closed 2023-02-26 by sascha988

virtadpt, 2023-03-23:
Wow!  This works when the existing Edit -> foo -> Import/Export function doesn't work!