Part-DB/Part-DB-server #228: Maximum file size

mase76, 2023-02-27:
Hi,
I think, the maximum filesize of 16M is to small for some datasheets. As far I have seen, this size is hardcoded in AttachmentFormType.php. Maybe you could make it configurable.

jbtronics, 2023-02-27:
Sounds reasonable. It should not be much work to make this configurable via config file. I guess something like 100 MB for the default value should be fine (and if more is really needed, admins can increase this limit).

I will try to include that in the next minor release (1.1.0)

jbtronics, 2023-03-02:
This is implemented in Part-DB 1.0.3

closed 2023-03-02 by jbtronics

CyberTeckyLLC, 2024-10-30:
As a suggestion, it would be helpful to reference how to increase the attachment 'Upload file' size by editing the php.ini configuration.  I believe the default file size is 2.00 MB (which may accommodate most avatars), however, for files such as datasheets (i.e., typically .pdf formats), are typically larger in size.

[image]

That being said, the .env.local in /var/www/partdb does reference the the php.ini file, however, the #comment did not necessarily mention where the PHP configuration file was located.  Granted, those who have PHP experience may know where to find this file, but for those novice web admins like myself, it would be great to see a reference somewhere in the .env.local file, for example:

/var/www/partdb/.env.local

[image]

Overall, PartDB is a great project and you definitely have gained a new $$$ contributor!  Keep up that great work...

Best regards,

CyberTecky

CyberTeckyLLC, 2024-10-30:
As a suggestion, it would be helpful to reference how to increase the attachment 'Upload file' size by editing the php.ini configuration.  I believe the default file size is 2.00 MB (which may accommodate most avatars), however, for files such as datasheets (i.e., typically .pdf formats), are typically larger in size.

[image]

Recommendation:  Add where to edit the php.ini configuration file (e.g., /etc/php/8.x/apache2/php.ini)

[image]

Overall, PartDB is a great project and you definitely have gained a new $$$ contributor!  Keep up that great work...

Best regards,

CyberTecky

tobiasfalk, 2024-12-29:
Hi I have changed my php.ini file to "upload_max_filesize =100M" and "post_max_size=100M" but I still can not upload pictures larger then 2M, I tried restarting everything but did not work.
Did I miss something?
Also when I type `php -r "echo ini_get('upload_max_filesize');"` I get`100M`, sugseting that on the PHP side everthing works