Part-DB/Part-DB-server #229: Align QR-code and text on a label

mase76, 2023-03-01:
Hi,
I am not able to align the text to the left when the QR-Code is enabled. See the attached image.
[image]

jbtronics, 2023-03-01:
I guess the margins or paddings of the elements are to large and he do not want to break the text inside a word. 

You could try to add to the additional styles field something like

```CSS

body {

word-wrap: break-word;
}

```

mase76, 2023-03-03:
This did not have any effect.

jbtronics, 2023-03-18:
What are the exact parameters you are using to get these issue (especially the label size?). Have you applied any special styling to the text?
I tried to replicate it with a fresh label, but could not really replicate the issue.

mase76, 2023-03-18:
Nothing special. Just enabled QR-code and 3 textitems with different sizes.. The label size is 45x36mm.
word-wrap: break-word; helped a little to get the full information on the label. But the space on the lower left is still there.