Part-DB/Part-DB-server #230: Check if object is its own parent before saving

suuppl, 2023-03-02:
**Describe the bug**
When editing an object, it is possible to select itself as its parent which seems to break the database (I had to reinstall part-db-server the last time I made that mistake). part-db shows "504 Gateway Time-out"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Edit Units" (or some other object)
2. Create an object
3. Save
4. Change the objects parent
5. Save

**Expected behavior**
Part-DB should throw an error. something like "a category can't be its own parent". Instead, something breaks.

**Screenshots**

**Server Side**
- Part-DB Version: Part-DB 1.0.1
- PHP Version: PHP 8.1.2
- Database Server: SQLite 3.37.2

jbtronics, 2023-03-02:
In theory that should already happen. But I will look into that (also it should not completely break when this happens)

suuppl, 2023-03-02:
I rescued it this time by moving the database somewhere else, generating a new one and moving the database back + clearing the cache. (I updated to 1.0.2 while i was at it.) Thanks for your work!

jbtronics, 2023-03-02:
As far as I could replicate it, the invalid parent will not be saved to the database (as the already existing checks are preventing that), so it should be enough to clear the cache (and maybe restart the webserver, to ensure the request has ended).

However this bug should now be fixed in Part-DB 1.0.3

closed 2023-03-02 by jbtronics