Part-DB/Part-DB-server #233: Rules for IPN

marcusgrtz, 2023-03-03:
It would be nice not to use the IPN as a free field, but to store a set of rules in the settings.

- Unique IPN names (yes/no)
- Automatically generate new IPN (yes/no)
- IPN prefix
- IPN suffix
- Start numbering from
- Numbering with leading zeros (yes/no)
- Number of leading zeros
- Only allow automatically generated IPN (yes/no)

zanna-37, 2023-03-05:
I would also love this feature!

jbtronics, 2023-03-21:
How does the format of the IPNs you use look like?
The problem is that there could be many different formats, which may not be configurable only with your described parameters.

marcusgrtz, 2023-03-21:
**Example 1 with settings:**

- Unique IPN names: yes
- Automatically generate new IPN: yes
- IPN prefix: ACME
- IPN suffix: 
- Start numbering from: 15
- Numbering with leading zeros: yes
- Number of leading zeros: 6
- Only allow automatically generated IPN: yes

The result for the next IPNs looks: ACME000015, ACME000016, ACME000017

**Example 2 with settings:**

- Unique IPN names: yes
- Automatically generate new IPN: yes
- IPN prefix: 
- IPN suffix: -PART 
- Start numbering from: 329
- Numbering with leading zeros: yes
- Number of leading zeros: 4
- Only allow automatically generated IPN: yes

The result for the next IPNs looks: 0329-PART, 0330-PART, 0331-PART

zanna-37, 2023-03-21:
It might be overkill, but a parametrized field would be amazing and allows having maximum flexibility.
E.g. `ACME-{counter}-PART`.
In this way, we could also use additional placeholders, similarly to what we already have in the label generator. The only novelty would be a general counter that the user set (start numbering, leading zeros, etc.).

Anyway, the example provided by Marcus would fit my needs as well.

jaumefe, 2024-08-05:
Any update on this topic?

I would also like to have this feature. For example, the way I use to organize my database is by the Category of the component.

The structure of the Internal Part Number that I am using is a 8 digit number in which the 2 first numbers represents the Category, for example:

- 10: IC
- 11: Capacitor
- 12: Resistor
- 13: Inductor
- etc

Then the other 6 numbers are just for identifying each component inside a Category.

I think a common practice is to group components by Category so maybe, the rules could be defined independent for every Category

If it's too complicated because of the variability on the pattern of everyone's IPN, it may be useful, at least in my case, to be able to see the latest IPN added to the corresponding Category

va7wv, 2025-12-29:
This would be great.  At the very least, it would be nice to choose the starting number for each category and sub.