Part-DB/Part-DB-server #234: One measurement unit as default

mase76, 2023-03-03:
Hi,
it would be nice, if one measurement unit could be set as default. The unit setting is often forgotten in the advanced option of a part.

jbtronics, 2023-03-07:
What is your use case for that? In general I would expect that the majority of parts are pieces and therefore have no measurement unit. And if you have a non-pieces part I would expect that you would need to choose in the most times between volumes, length, etc.

mase76, 2023-03-07:
I've created some units, also "pieces". Ok, I could remove the unit "pieces" and take an empty unit field as pieces.

Schonke, 2024-01-24:
Seconding this suggestion. 
I'm using a category for "consumables" with sub-categories for tracking things like solder, flux and filament for the 3D printer.

Having the unit default to, for example, grams for every item in the category "filament" would be really nice and save a couple of extra clicks for each new item added to the database.

henrikkoch, 2024-01-24:
I agree on that suggestion. In that way you can have an overview of all the printer filament that you have laying around. With tags like material (ABS, ASA, PLA etc. ) and colors, manufactures, etc.