Part-DB/Part-DB-server #237: thumbnails are not shown correctly

001oc, 2023-03-06:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Server Side**
- Part-DB Version: 1.0.3
- PHP Version: PHP 8.1.16
- Database Server MySQL 8.0.32

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Firefox
 - Version 110.0

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

I have created some parts together with sone cathegories and footprints and everything seems to be working great for some day. Since two days the thumbnails are not working anymore.

Thumbnails (all, not only some) are not shown correctly as one can see at the attachted screenshots. Ich have tried to recreate footprints and parts which doesn't help. I have done cache clearing and the db migrate process. The containers (partdb and mysql) are restartet more than onces. When I right click the text that comes up instead of the thumbnail to open the thumbnail in a seperate tab the pircture is displayed without any problem in its normal size (not thumnailed). As of that, the pircture to produce the thumbnails is obviously existing but the conversion to a thumbnail is not working correctly I think.

What can I do to get it working again.
[image]
[image]
[image]

jbtronics, 2023-03-06:
What PHP extensions have you installed? You can see that with `php -m`.

Maybe the gd library used to generate the thumbnails are missing.

001oc, 2023-03-06:
Thanks for the reply. Here is what php -m says. Seems that gd is enabled.

[PHP Modules]
bcmath
calendar
Core
ctype
curl
date
dom
exif
FFI
fileinfo
filter
ftp
gd
gettext
hash
iconv
intl
json
libxml
mbstring
mysqli
mysqlnd
openssl
pcntl
pcre
PDO
pdo_mysql
pdo_sqlite
Phar
posix
readline
Reflection
session
shmop
SimpleXML
sockets
sodium
SPL
sqlite3
standard
sysvmsg
sysvsem
sysvshm
tokenizer
xml
xmlreader
xmlwriter
xsl
Zend OPcache
zip
zlib

[Zend Modules]
Zend OPcache

jbtronics, 2023-03-07:
That only the img alt text is shown, looks like the browser can not load the thumbnails from the server.
Do you get any error messages in the browsers developer console, that it can not load the ressources?

Have you checked the error logs of your server? When using the default Part-DB docker image, apache error should be visible in the stderr of the docker container (so with docker logs). Do you see any error messages there when trying to access the page?

Do you have a `public/media/cache/` folder in your Part-DB instance? What are the access permissions of it?
You could try to remove this folder (thats where the created thumbnails are saved), which forces the generation of new ones.

MG-5, 2023-03-26:
Sadly i have this issue too. Trying to clear cache and delete the public_media folder only lead to error 500 with the consequence i must recreate the docker container - restarting it is not sufficient.
The folders has the permission www-data:www-data.

There are no errors in the logs

MG-5, 2023-03-26:
Ah, showing in Firefox Browser Dev Console i got the cause of the problem.

It gives an error  for every thumbnail like this:
`Content Security Policy: Unsichere Anfrage 'http://partdb.****.de/media/cache/thumbnail_sm/img/footprints/Passive/Capacitors/Electrolyte/Radial/Electrolyte_RADIAL_2RM_5D.png' wird blockiert.`

Which means, that in https environment, all http-requests were blocked. When I call partdb in local network (without https) the thumbnails were loaded correctly.

001oc, 2023-03-27:
When I had that problem the other day I had different outcomes with Firefox and Edge. Firefox shows the described failures while Edge was still working correctly ... looked for me like a cache problem -> php bin/console cache:clear helped me .

Currently I have it running smoothly except some "500 Internal Server Error" which I am currently "solving" with restarting my containers respectivetly.

MG-5, 2023-03-27:
Clearing cache and restarting has no effect for me.

jbtronics, 2023-03-27:
> Ah, after looking in Firefox Browser Dev Console i got the cause of the problem.
> 
> It gives an error  for every thumbnail like this:
> `Content Security Policy: Unsichere Anfrage 'http://partdb.****.de/media/cache/thumbnail_sm/img/footprints/Passive/Capacitors/Electrolyte/Radial/Electrolyte_RADIAL_2RM_5D.png' wird blockiert.`
> 
> Which means, that in https environment, all http-requests will be blocked. When I call partdb in local network (without https) the thumbnails were loaded correctly.

Seems plausible. I guess there is somewhere an absolute url including the procotol generated, which causing this issue and Part-DB can not determine that it is accessed via HTTPS (this should happen, when you using an Reverse Proxy).

Are you using a reverse proxy and if so, have you set the trusted proxies setting?

MG-5, 2023-03-27:
I am using Nginx as reverse proxy, better the docker image "swag" from linuxserver.io.
Yes, i setted `TRUSTED_PROXIES` to `swag` (ip address instead of container name doesnt matter)

In the Nginx configuration it also says that all http requests are automatically redirected to https. If I go to the http-link from Firefox dev console, then it automatically jumps to an https link. 

If I go to the overview of the respective part, the footprint image is transferred correctly via https.
Only the thumbnails in the list are not.

jbtronics, 2023-03-31:
With the latest commit which can be found in the master branch, thumbnails are network relative pathes (so without schema specification). This should fix this issue.

closed 2023-03-31 by jbtronics

MG-5, 2023-04-01:
Pulled the latest docker image from master tag and now its working!
Thanks you!