Part-DB/Part-DB-server #240: I can not get it working on Docker on synology NAS

Tallen-t, 2023-03-09:
I get a database error. 

[image]

jbtronics, 2023-03-09:
You have to run the database migration command inside your docker container to initialize the database.

See step 3 here: https://docs.part-db.de/installation/installation_docker.html

Basically you have to run `docker exec --user=www-data partdb php bin/console doctrine:migrations:migrate`, where partdb is the name of your docker container.

Tallen-t, 2023-03-09:
ok. Got it going but now cant see any parts that i have created.\
Nothing in logs
[image]
[image]

Tallen-t, 2023-03-10:
6 hours later....
I still can not view my parts, and i noticed that the same problem is occuring on the demo website.
The demo website shows almost a thousand parts but none are viewable

jbtronics, 2023-03-10:
That's issue #241 
Discussion will be continued there

closed 2023-03-10 by jbtronics