Part-DB/Part-DB-server #241: Cannot see any parts after browser cache clear

mase76, 2023-03-09:
Hi,
since I cleared my browser cache, I don't see any parts. Everything else, like categories, footprints, etc. is visible. Under "show statistics" I see the correct numer of parts. I cannot see anything in the logs. I am on Debian 11 with Mariadb. The PHP version i 7.4.

mase76, 2023-03-09:
I did a downgrade to 1.0.3 and the parts are shown again as expected.

mase76, 2023-03-09:
Seems that 02134dc959a27fc355a257b57e69c55695b3cef9 causes the problem.

jbtronics, 2023-03-10:
> Seems that [02134dc](https://github.com/Part-DB/Part-DB-server/commit/02134dc959a27fc355a257b57e69c55695b3cef9) causes the problem.

Yeah this seems plausible. If no entry was set data is null, and trying to set anything on it will cause an error in the Browser which fails to load the table.
In my testings I had some old data set, so I didn't notice...

Should be easy to fix.

Tallen-t, 2023-03-10:
working on demo site.
can you update docker image?

mase76, 2023-03-10:
I can confirm, that this issue is fixed in v1.1.1.

closed 2023-03-11 by jbtronics