Part-DB/Part-DB-server #256: JavaScript TypeError from a null value, causes tables not to display.

AmuletIndustries, 2023-03-23:
**Describe the bug**
On some browsers on some machines, when loading any page with a table populated by data from the database, a null value is returned during the table load and a JavaScript Uncaught TypeError occurs. This error occurs in Chrome, Edge, and Firefox on some Win10 machines, but works correctly in Chrome on other Win10 machines. The error occurs in Chrome on Android and iOS, as well as Safari on iOS. On at least one machine the page loads correctly in Chrome but experiences this error in Firefox. Even on browsers where this error occurs, pages which modify the database without displaying it in a table (like part creation, category creation, etc.) work normally and the database is modified as expected. The error occurs regardless of which user is signed in, and the anon user has permissions allowing it to see things that this error breaks.

Importantly: While it seems to be arbitrary if a given browser/machine has this error, the error is persistent on that browser and no amount of restarting or refreshing or changing settings causes that to change.

**To Reproduce**
Steps to reproduce the behavior:
 - Difficult to reproduce, inconsistent in what machines will/won't work.
 - On machines where this error occurs:
1. Open browser.
2. Navigate to Part-DB IP.
3. Attempt to load any page with a table, such as the See All Parts page, or the contents of any category.
4. If the error is occurring, the table will be completely empty. Inspecting the browser console for errors will show an `Uncaught TypeError: e is null`

**Expected behavior**
The table on a given page should load normally.

**Screenshots**
All of these screenshots were taken on the same Win10 machine minutes from each other accessing the same page on the same server.

[image]
The above image is a screenshot of Chrome showing the Show All Parts table correctly populated.

[image]
The above image is a screenshot of the same page on the same server and the same client Win10 machine but in Firefox. Note the empty table.

[image]
The above image is the console output from the inspection, showing the error as well as the offending line in the JS file.

**Server Side**
- Part-DB Version: 1.1.0
- PHP Version: 8.1.16
- Database Server sqlite 3.34.1

**Desktop (please complete the following information):**
 - OS: Windows 10 Enterprise 22H2
 - Browser Works in Chrome, Fails in Firefox
 - Chrome: Version 111.0.5563.65 (Official Build) (64-bit)
 - Firefox: Version 102.9.0esr (64-bit)

**Smartphone (please complete the following information):**
 - Device: Pixel 6 Pro
 - OS: Android 13, Security Update March 5, 2023
 - Browser Chrome (error occurs on this Chrome install)
 - Version 111.0.5563.115

jbtronics, 2023-03-23:
This is issue #241. This was already fixed on Version 1.1.1

If this still occurs after you have upgraded to the latest version, feel free to open the issue again.

closed 2023-03-23 by jbtronics

AmuletIndustries, 2023-03-23:
Ah I thought I had checked through the issues list but I clearly wasn't thorough enough. Updating to the latest version fixes it. Thank you very much.