Part-DB/Part-DB-server #257: Behaviour of the minimal stock

mase76, 2023-04-03:
Hi!
What is the current behaviour of the minimal stock field of a part? Can I get an overview somewhere, where the parts, which fell below the minimal stock are listed?

jbtronics, 2023-04-07:
Now, when the total amount is smaller than the minimal amount, the field will get marked red for highlighting.
Also, you can now filter for parts where  this is the case (or where this is not the case), with the newly added constraint "In stock less than desired (total amount < min. amount)" in the stocks tab in part lists filter.

closed 2023-04-07 by jbtronics

Pyromane, 2023-04-11:
Show > Show all Parts
Filter > Stock > In stock less than desired (total amount < min. amount)

mase76, 2023-09-03:
Wouldn't a notification on the start page be better? A notification, that one or more parts are under the minimal stock would be enough.

PC-maniak, 2024-12-25:
Hi, I am currently using part-db for private purposes and I agree with "mase76" -> please provide a warning on the (Main) banner of the page that this or that component is below the limit..

You simply forget to order and every time you log in you will be reminded. =) please ..