Part-DB/Part-DB-server #258: Deleting category

blue3121, 2023-04-04:
404 error when deleting a category

**To Reproduce**
Delete an existing category

**Expected behavior**
Perhaps you can bring up a category entry page

**Server Side**
- Part-DB Version: 1.3.0-dev (master/749e7db) 
- PHP Version: php 8.1.2
- Database Server sqlite

**Desktop (please complete the following information):**
 - OS: Kubuntu 22.04
 - Browser Firefox 111.0.1

jbtronics, 2023-04-04:
Interesting, this seems only to happen when using Firefox. Even in the demo you can see some error message (I guess with the 404) popup, which does not show up when using Chrome (and which is not intended). Sometimes you get redirected to the old page (which then 404) instead of the correct new element page. I guess this somehow depends on the server timing, so you maybe see it more often than on the demo.

My first guess is, that is has something to do with the recently changed error popup mechanism. I will look into that.

blue3121, 2023-04-04:
I solved. i cleared the yarn cache directory and re-installed.
I hadn't noticed an error during the yarn install and build.
I'm sorry.
Now it works !
Thank you
My best regards

Prt-DB is GREAT !!!

closed 2023-04-04 by blue3121

blue3121, 2023-04-04:
I solved. I cleared the yarn cache directory and re-installed.
I hadn't noticed an error during the yarn install and build.
I'm sorry.
Now it works !
Thank you
My best regards

Part-DB is GREAT !!!