Part-DB/Part-DB-server #260: DB migration issue

Pyromane, 2023-04-08:
I am running a Part-DB-legacy version: 0.5.15 (stable) and trying to migrate to the latest Git Master version of Part-DB-server and getting the following error message when trying to migrate:
`php bin/console doctrine:migrations:migrate

 WARNING! You are about to execute a migration in database "usr_web483_1" that could result in schema changes and data loss. Are you sure you wish to continue? (yes/no) [yes]:
 >

[notice] Migrating up to DoctrineMigrations\Version20230220221024
[error] Migration DoctrineMigrations\Version20190902140506 failed during Execution. Error: "An exception occurred while executing a query: SQLSTATE[42S02]: Base table or view not found: 1146 Table 'usr_web483_1.attachements' doesn't exist"
01:32:47 CRITICAL  [console] Error thrown while running command "doctrine:migrations:migrate". Message: "An exception occurred while executing a query: SQLSTATE[42S02]: Base table or view not found: 1146 Table 'usr_web483_1.attachements' doesn't exist" ["exception" => Doctrine\DBAL\Exception\TableNotFoundException^ { …},"command" => "doctrine:migrations:migrate","message" => "An exception occurred while executing a query: SQLSTATE[42S02]: Base table or view not found: 1146 Table 'usr_web483_1.attachements' doesn't exist"] ["token" => null,"command" => ["name" => "doctrine:migrations:migrate","arguments" => ["command" => "doctrine:migrations:migrate","version" => "latest"]]]

In ExceptionConverter.php line 49:

  An exception occurred while executing a query: SQLSTATE[42S02]: Base table or view not found: 1146 Table 'usr_web483_1.attachements' doesn't exist

In Exception.php line 28:

  SQLSTATE[42S02]: Base table or view not found: 1146 Table 'usr_web483_1.attachements' doesn't exist

In Connection.php line 70:

  SQLSTATE[42S02]: Base table or view not found: 1146 Table 'usr_web483_1.attachements' doesn't exist

doctrine:migrations:migrate [--write-sql [WRITE-SQL]] [--dry-run] [--query-time] [--allow-no-migration] [--all-or-nothing [ALL-OR-NOTHING]] [--configuration CONFIGURATION] [--em EM] [--conn CONN] [--] [<version>]`

But the table is present in the database (=MariaDB version 10.5.18)
``CREATE TABLE `attachements` (
  `id` int(11) NOT NULL,
  `name` tinytext NOT NULL,
  `class_name` varchar(255) NOT NULL,
  `element_id` int(11) NOT NULL,
  `type_id` int(11) NOT NULL,
  `filename` mediumtext NOT NULL,
  `show_in_table` tinyint(1) NOT NULL DEFAULT 0,
  `last_modified` timestamp NOT NULL DEFAULT '0000-00-00 00:00:00'
) ENGINE=InnoDB DEFAULT CHARSET=utf8 COLLATE=utf8_unicode_ci;``

`INSERT INTO `attachements` (`id`, `name`, `class_name`, `element_id`, `type_id`, `filename`, `show_in_table`, `last_modified`) VALUES
(1, 'Bild', 'Part', 14, 1, '%BASE%/data/media/img_0559e001bbbee3fc82869ffb38bfe40b.jpg', 0, '0000-00-00 00:00:00'),`

I tried to paste the corresponding error messages as code, but unfortunately it doesn't work at all .

jbtronics, 2023-04-08:
Have you run the migration command already, sometime before you copied this error message? The error can change on subsequent runs, when the database was already modified by earlier failed migration attempts.

I think the easiest way to debug this, would be if you could send me a SQL dump of your database (the backuped version, before you tried to migrate).

Pyromane, 2023-04-08:
**Thanks for the tip, to test I have now done the following:
Deleted table, restored from backup and started migration:** 
`php bin/console doctrine:migrations:migrate

 WARNING! You are about to execute a migration in database "usr_web483_1" that could result in schema changes and data loss. Are you sure you wish to continue? (yes/no) [yes]:
 >

[notice] Migrating up to DoctrineMigrations\Version20230220221024
[warning] Old Part-DB Database detected! Continue with upgrade...
[warning] Migration DoctrineMigrations\Version1 was executed but did not result in any SQL statements.
[error] Migration DoctrineMigrations\Version20190902140506 failed during Execution. Error: "An exception occurred while executing a query: SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP FOREIGN KEY `parts_id_storelocation_fk`; check that it exists"
18:30:51 CRITICAL  [console] Error thrown while running command "doctrine:migrations:migrate". Message: "An exception occurred while executing a query: SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP FOREIGN KEY `parts_id_storelocation_fk`; check that it exists" ["exception" => Doctrine\DBAL\Exception\DriverException^ { …},"command" => "doctrine:migrations:migrate","message" => "An exception occurred while executing a query: SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP FOREIGN KEY `parts_id_storelocation_fk`; check that it exists"] ["token" => null,"command" => ["name" => "doctrine:migrations:migrate","arguments" => ["command" => "doctrine:migrations:migrate","version" => "latest"]]]

In ExceptionConverter.php line 117:

  An exception occurred while executing a query: SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP FOREIGN KEY `parts_id_storelocation_fk`; check that it exists

In Exception.php line 28:

  SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP FOREIGN KEY `parts_id_storelocation_fk`; check that it exists

In Connection.php line 70:

  SQLSTATE[42000]: Syntax error or access violation: 1091 Can't DROP FOREIGN KEY `parts_id_storelocation_fk`; check that it exists

doctrine:migrations:migrate [--write-sql [WRITE-SQL]] [--dry-run] [--query-time] [--allow-no-migration] [--all-or-nothing [ALL-OR-NOTHING]] [--configuration CONFIGURATION] [--em EM] [--conn CONN] [--] [<version>]`

[usr_web483_1.zip](https://github.com/Part-DB/Part-DB-server/files/11183936/usr_web483_1.zip)
**The table structure of the backup can be found in the appendix.
If you need the full backup, please let me know how to send it to you. Thanks a lot!**

jbtronics, 2023-04-08:
For some reason, your database is completely missing every foreign key,  even though it has the indices, which should have been created in the same migration step in legacy Part-DB. The new Part-DB deletes this keys to modify them, which caused this exception when the FK were not already existing.

Now it is checked that the foreign keys are existing before trying to delete them and the migration should run smoothly. After the migration, your database should have the same structure as everybody else.

Feel free to open the issue again, if you still encounter issues (Ensure that you use the backup database, which was not touched by any migration attempt)

closed 2023-04-08 by jbtronics

Pyromane, 2023-04-10:
The migration could now be completed successfully. Many thanks for the support!