Part-DB/Part-DB-server #261: TypeError when accessing history (maybe migration problem)

sunflower-seed, 2023-04-08:
**Describe the bug**

I ugraded my PartDB-legacy installation successfully, using the given tutorials.

After logging in or when i try to access the log i encounter a 500 error.
It might be an error during migration or a general error.
In either way it should be fixed.

**Analysis**
I presume some old PartDB-legacy Version wrote not string values to the "i" Field in the extra JSON in the DB.

**Debugging and workaround**

I did a top to bottom debugging and fixing

1. Server logs

`{"message":"Uncaught Error: App\\Entity\\LogSystem\\ElementCreatedLogEntry::getCreationInstockValue(): Return value must be of type ?string, int returned","context":{"exception":{"class":"TypeError","message":"App\\Entity\\LogSystem\\ElementCreatedLogEntry::getCreationInstockValue(): Return value must be of type ?string, int returned","code":0,"file":"partdb/src/Entity/LogSystem/ElementCreatedLogEntry.php:57"}},"level":500,"level_name":"CRITICAL","channel":"php","datetime":"2023-04-08T00:00:00.605610+02:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"user","username":"user"},"url":"/de/log/","ip":"192.168.1.1","http_method":"POST","server":"domain.invalid","referrer":"https://domain.invalid/de/log/"}}
{"message":"Uncaught PHP Exception TypeError: \"App\\Entity\\LogSystem\\ElementCreatedLogEntry::getCreationInstockValue(): Return value must be of type ?string, int returned\" at partdb/src/Entity/LogSystem/ElementCreatedLogEntry.php line 57","context":{"exception":{"class":"TypeError","message":"App\\Entity\\LogSystem\\ElementCreatedLogEntry::getCreationInstockValue(): Return value must be of type ?string, int returned","code":0,"file":"partdb/src/Entity/LogSystem/ElementCreatedLogEntry.php:57"}},"level":500,"level_name":"CRITICAL","channel":"request","datetime":"2023-04-08T00:00:00.606703+02:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"user","username":"phil"},"url":"/de/log/","ip":"192.168.1.1","http_method":"POST","server":"domain.invalid","referrer":"https://domain.invalid/de/log/"}}
`

2. Database 

I searched the DB and spotted some entries that having some string values, that might cause the problem:
` 
SELECT * FROM `log` WHERE id=1345 OR id=1352;
+------+---------------------+---------+-------+------+-----------+-------------+---------+----------+
| id   | datetime            | id_user | level | type | target_id | target_type | extra   | username |
+------+---------------------+---------+-------+------+-----------+-------------+---------+----------+
| 1345 | 2022-07-29 11:49:37 |       2 |     6 |    6 |       169 |          10 | {"i":1} | user     |
| 1352 | 2022-08-10 17:10:15 |       2 |     6 |    6 |       170 |          10 | {"i":4} | user     |
+------+---------------------+---------+-------+------+-----------+-------------+---------+----------+
`
3. Quick Fix

In [ElementCreatedLogEntry](https://github.com/Part-DB/Part-DB-server/blob/master/src/Entity/LogSystem/ElementCreatedLogEntry.php#L57) change the return value of 'getCreationInstockValue(): ?string'

` return isset($this->extra['i']) ? strval($this->extra['i']) : null;`

**Server Side**
- Part-DB Version: Part-DB 1.2.0
- PHP Version: [e.g. PHP 7.4.3]
- Database Server : MariaDB 10.6.12

jbtronics, 2023-04-08:
Thanks for reporting.

I added the fix like you suggested it. Fixing the JSON in the database itself is too complicated and has no real advantage.
Now should be fixed for everybody.

closed 2023-04-08 by jbtronics

Pyromane, 2023-04-10:
The migration could now be completed successfully. Many thanks for the support!