Part-DB/Part-DB-server #263: After upgrading from legacy version, I can't log in.

Pyromane, 2023-04-10:
Hello,

I had been using the legacy version almost entirely as an anonymous user and I see this content even without logging in.

When I try to log in as admin, no error message comes up, no matter if I enter the valid admin login or freely invented data, but always end up back at the login screen.

By using
```
php bin/console partdb:users:set-password admin
php bin/console cache:clear
```
I already tried to set a new PW and clear the cache.

/var/log/prod.log does not exist for me, although I have a native installation
/var/log/apache2/partdb_error.log contains no content.
/var/log/apache2/partdb_access.log 
```192.168.0.11 - - [10/Apr/2023:13:23:29 +0000] "POST /en/login HTTP/1.1" 302 1004 "http://192.168.0.18/de/login" "Mozilla/5.0 (Windows NT 6.1; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/109.0.0.0 Safari/537.36 OPR/95.0.0.0"```
The developer console in the browser doesn't provide great insights from my point of view either: 
[image]

How can I get more log messages?
Thanks a lot!

Pyromane, 2023-04-10:
```
 php bin/console partdb:users:list

[INFO] Found 2 users in database.
Users:
======
+----+-----------+------+-------+----------------+----------------+-------+
| ID | Username  | Name | Email | Group          | Login Disabled | Type  |
+----+-----------+------+-------+----------------+----------------+-------+
| 1  | anonymous |      |       | users (ID: 3)  | No             | Local |
| 2  | admin     |      |       | admins (ID: 1) | No             | Local |
+----+-----------+------+-------+----------------+----------------+-------+
```

jbtronics, 2023-04-11:
> /var/log/prod.log does not exist for me, although I have a native installation

You must not look into the absolute path (so in your file system), but into the `var/log` folder inside your Part-DB installation. There you should find more detailed error messages.

IvanBayan, 2023-04-14:
@Pyromane have you tried to execute `php bin/console partdb:users:upgrade-permissions-schema`?

Pyromane, 2023-04-15:
> @Pyromane have you tried to execute `php bin/console partdb:users:upgrade-permissions-schema`?

Thanks for the tip, that was the solution to the problem!

jbtronics, 2023-04-16:
Im actually surprised (and can't really explain why), that this command did change anything on your problem.

Normally the permission schema migration is done on demand in the next request of a user (so in your case the login). This command is not really needed, it's more intended for debug purposes.

Also your output suggests that nothing on your admin user were changed, so I don't really know what happened.

However I am glad to hear, that it works now.

Pyromane, 2023-04-16:
My attempts to recreate the whole thing failed, now the migration command is perfectly sufficient and I can log in with my admin password immediately afterwards.

Thanks for your support jbtronics and IvanBayan!

closed 2023-04-16 by Pyromane