Part-DB/Part-DB-server #271: Legacy upgrade attempt using Docker

Pyromane, 2023-04-12:
I started another independent migration attempt, for this I chose the Docker installation including MySQL. I made the MySQL installation port accessible and imported the data from the legacy installation.

I use the same export as the last attempt, so I link the same export: https://github.com/Part-DB/Part-DB-server/files/11183936/usr_web483_1.zip

When I try to migrate, I get the following error message:
```php bin/console doctrine:migrations:migrate

 WARNING! You are about to execute a migration in database "partdb" that could result in schema changes and data loss. Are you sure you wish to continue? (yes/no) [yes]:
 > 

[notice] Migrating up to DoctrineMigrations\Version20230408213957
[warning] Old Part-DB Database detected! Continue with upgrade...
[warning] Migration DoctrineMigrations\Version1 was executed but did not result in any SQL statements.
[error] Migration DoctrineMigrations\Version20190902140506 failed during Execution. Error: "An exception occurred while executing a query: SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'groups' at line 1"
{"message":"Error thrown while running command \"doctrine:migrations:migrate\". Message: \"An exception occurred while executing a query: SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'groups' at line 1\"","context":{"exception":{"class":"Doctrine\\DBAL\\Exception\\SyntaxErrorException","message":"An exception occurred while executing a query: SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'groups' at line 1","code":1064,"file":"/var/www/html/vendor/doctrine/dbal/src/Driver/API/MySQL/ExceptionConverter.php:86","trace":["/var/www/html/vendor/doctrine/dbal/src/Connection.php:1931","/var/www/html/vendor/doctrine/dbal/src/Connection.php:1874","/var/www/html/vendor/doctrine/dbal/src/Connection.php:1099","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:298","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:173","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:96","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:107","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:71","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:157","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Tools/Console/Command/MigrateCommand.php:211","/var/www/html/vendor/symfony/console/Command/Command.php:298","/var/www/html/vendor/symfony/console/Application.php:1058","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:96","/var/www/html/vendor/symfony/console/Application.php:301","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:82","/var/www/html/vendor/symfony/console/Application.php:171","/var/www/html/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:54","/var/www/html/vendor/autoload_runtime.php:35","/var/www/html/bin/console:11"],"previous":{"class":"Doctrine\\DBAL\\Driver\\PDO\\Exception","message":"SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'groups' at line 1","code":1064,"file":"/var/www/html/vendor/doctrine/dbal/src/Driver/PDO/Exception.php:28","trace":["/var/www/html/vendor/doctrine/dbal/src/Driver/PDO/Connection.php:75","/var/www/html/vendor/doctrine/dbal/src/Connection.php:1094","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:298","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:173","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:96","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:107","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:71","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:157","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Tools/Console/Command/MigrateCommand.php:211","/var/www/html/vendor/symfony/console/Command/Command.php:298","/var/www/html/vendor/symfony/console/Application.php:1058","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:96","/var/www/html/vendor/symfony/console/Application.php:301","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:82","/var/www/html/vendor/symfony/console/Application.php:171","/var/www/html/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:54","/var/www/html/vendor/autoload_runtime.php:35","/var/www/html/bin/console:11"],"previous":{"class":"PDOException","message":"SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'groups' at line 1","code":42000,"file":"/var/www/html/vendor/doctrine/dbal/src/Driver/PDO/Connection.php:70","trace":["/var/www/html/vendor/doctrine/dbal/src/Driver/PDO/Connection.php:70","/var/www/html/vendor/doctrine/dbal/src/Connection.php:1094","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:298","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:173","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:96","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:107","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:71","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:157","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Tools/Console/Command/MigrateCommand.php:211","/var/www/html/vendor/symfony/console/Command/Command.php:298","/var/www/html/vendor/symfony/console/Application.php:1058","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:96","/var/www/html/vendor/symfony/console/Application.php:301","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:82","/var/www/html/vendor/symfony/console/Application.php:171","/var/www/html/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:54","/var/www/html/vendor/autoload_runtime.php:35","/var/www/html/bin/console:11"]}}},"command":"doctrine:migrations:migrate","message":"An exception occurred while executing a query: SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'groups' at line 1"},"level":500,"level_name":"CRITICAL","channel":"console","datetime":"2023-04-12T21:58:12.370788+00:00","extra":{"token":null,"command":{"name":"doctrine:migrations:migrate","arguments":{"command":"doctrine:migrations:migrate","version":"latest"}}}}
21:58:12 CRITICAL  [console] Error thrown while running command "doctrine:migrations:migrate". Message: "An exception occurred while executing a query: SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'groups' at line 1" ["exception" => Doctrine\DBAL\Exception\SyntaxErrorException^ { …},"command" => "doctrine:migrations:migrate","message" => "An exception occurred while executing a query: SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'groups' at line 1"] ["token" => null,"command" => ["name" => "doctrine:migrations:migrate","arguments" => ["command" => "doctrine:migrations:migrate","version" => "latest"]]]
[... cut ...]
   the right syntax to use near 'groups' at line 1                                                                                                                  
                                                                                                                                                                    

doctrine:migrations:migrate [--write-sql [WRITE-SQL]] [--dry-run] [--query-time] [--allow-no-migration] [--all-or-nothing [ALL-OR-NOTHING]] [--configuration CONFIGURATION] [--em EM] [--conn CONN] [--] [<version>]```

jbtronics, 2023-04-14:
Should be fixed in latest master branch version

Pyromane, 2023-04-15:
I have changed the image to the following
`image: jbtronics/part-db1:master`
, deleted the database and restored the backup.
Now when I try to migrate I get the following error message:

```
php bin/console doctrine:migrations:migrate

 WARNING! You are about to execute a migration in database "partdb" that could result in schema changes and data loss. Are you sure you wish to continue? (yes/no) [yes]:
 > 

[notice] Migrating up to DoctrineMigrations\Version20230408213957
[warning] Old Part-DB Database detected! Continue with upgrade...
[warning] Migration DoctrineMigrations\Version1 was executed but did not result in any SQL statements.
[error] Migration DoctrineMigrations\Version20220925162725 failed during Execution. Error: "An exception occurred while executing a query: SQLSTATE[HY000]: General error: 1832 Cannot change column 'type_id': used in a foreign key constraint 'attachements_type_id_fk'"
{"message":"Error thrown while running command \"doctrine:migrations:migrate\". Message: \"An exception occurred while executing a query: SQLSTATE[HY000]: General error: 1832 Cannot change column 'type_id': used in a foreign key constraint 'attachements_type_id_fk'\"","context":{"exception":{"class":"Doctrine\\DBAL\\Exception\\DriverException","message":"An exception occurred while executing a query: SQLSTATE[HY000]: General error: 1832 Cannot change column 'type_id': used in a foreign key constraint 'attachements_type_id_fk'","code":1832,"file":"/var/www/html/vendor/doctrine/dbal/src/Driver/API/MySQL/ExceptionConverter.php:117","trace":["/var/www/html/vendor/doctrine/dbal/src/Connection.php:1931","/var/www/html/vendor/doctrine/dbal/src/Connection.php:1874","/var/www/html/vendor/doctrine/dbal/src/Connection.php:1099","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:298","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:173","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:96","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:107","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:71","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:157","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Tools/Console/Command/MigrateCommand.php:211","/var/www/html/vendor/symfony/console/Command/Command.php:298","/var/www/html/vendor/symfony/console/Application.php:1058","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:96","/var/www/html/vendor/symfony/console/Application.php:301","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:82","/var/www/html/vendor/symfony/console/Application.php:171","/var/www/html/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:54","/var/www/html/vendor/autoload_runtime.php:35","/var/www/html/bin/console:11"],"previous":{"class":"Doctrine\\DBAL\\Driver\\PDO\\Exception","message":"SQLSTATE[HY000]: General error: 1832 Cannot change column 'type_id': used in a foreign key constraint 'attachements_type_id_fk'","code":1832,"file":"/var/www/html/vendor/doctrine/dbal/src/Driver/PDO/Exception.php:28","trace":["/var/www/html/vendor/doctrine/dbal/src/Driver/PDO/Connection.php:75","/var/www/html/vendor/doctrine/dbal/src/Connection.php:1094","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:298","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:173","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:96","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:107","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:71","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:157","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Tools/Console/Command/MigrateCommand.php:211","/var/www/html/vendor/symfony/console/Command/Command.php:298","/var/www/html/vendor/symfony/console/Application.php:1058","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:96","/var/www/html/vendor/symfony/console/Application.php:301","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:82","/var/www/html/vendor/symfony/console/Application.php:171","/var/www/html/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:54","/var/www/html/vendor/autoload_runtime.php:35","/var/www/html/bin/console:11"],"previous":{"class":"PDOException","message":"SQLSTATE[HY000]: General error: 1832 Cannot change column 'type_id': used in a foreign key constraint 'attachements_type_id_fk'","code":0,"file":"/var/www/html/vendor/doctrine/dbal/src/Driver/PDO/Connection.php:70","trace":["/var/www/html/vendor/doctrine/dbal/src/Driver/PDO/Connection.php:70","/var/www/html/vendor/doctrine/dbal/src/Connection.php:1094","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:298","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:173","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/Version/DbalExecutor.php:96","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:107","/var/www/html/vendor/doctrine/migrations/lib/Doctrine/Migrations/DbalMigrator.php:71","/var/www/html/vendor/doctrin [... cut ...]
  SQLSTATE[HY000]: General error: 1832 Cannot change column 'type_id': used in a foreign key constraint 'attachements_ty  
  pe_id_fk'                                                                                                               
                                                                                                                          

doctrine:migrations:migrate [--write-sql [WRITE-SQL]] [--dry-run] [--query-time] [--allow-no-migration] [--all-or-nothing [ALL-OR-NOTHING]] [--configuration CONFIGURATION] [--em EM] [--conn CONN] [--] [<version>]
```

closed 2023-04-23 by jbtronics