Part-DB/Part-DB-server #272: Missing attachments after legacy migration

Pyromane, 2023-04-15:
The migration was successful without any error messages, but I am missing a lot of attachments.

This is already visible in the statistics:
**Legacy Version: 0.5.15 (stable)**
Number of attachments: 1533

**Version: 1.3.0-dev (master/e0e5fb3)**
Number of all file attachments 765
Number of all attachments uploaded by the user 603

The following entries can be read in the event log in PartD-DB:
[image]

While investigating the cause, I came across the following in the attachments table:
[image]
For the part with ID 334, the two images are displayed cleanly, but for item 335, neither the image nor the two PDFs are displayed.
Is this possibly related to the different "class_name"?

jbtronics, 2023-04-16:
Strange... I will look into it

jbtronics, 2023-04-17:
It was actually related to the wrong class_name. My own database missed a few attachments too, but never noticed it. So thanks for reporting.

With the newest commit there is a new database migration (run `php bin/console doctrine:migrations:migrate`), which fixes the class_name to the new format. Then the attachments should show up.

If you have deleted some parts since the migration to the Part-DB 1.0 version, and they had attachments associated, which had not shown up, they will be deleted too. (to prevent issues with database integrity)

closed 2023-04-17 by jbtronics