Part-DB/Part-DB-server #275: Special  characters

blue3121, 2023-04-20:
Where are greek alphabet symbols (like "omega"  "alpha" and so on) in the list of "notes" ?

Can you add the list also in "Unit symbol" in the "Measurement Unit" section?
The Greek alphabet is fundamental for most of the parameters and formulas used !

Thank You !!!

jbtronics, 2023-04-20:
I thought greek characters were part of the special symbols list in CKEditor, but it doesnt seems so. Will look into it.

You can type out the most commonly used greek characters on your keyboard in every Part-DB text field by pressing Alt + [Key]. 
For example to make a omega character by pressing Alt + W.
However many of these Alt keys are already assigned to some browser shortcuts, so i will maybe change the key to something else...

jbtronics, 2023-04-23:
The CKEDITOR special characters plugin now has all greek characters.

Besides that, the special character input was completely reworked and improved and allows you to input all greek characters and some other mathematical symbols in all text fields of Part-DB. See https://docs.part-db.de/usage/keybindings.html#special-characters

closed 2023-04-23 by jbtronics

jbtronics, 2023-04-24:
For Part-DB 1.3.1 the keybindings were changed a bit (for a more logical system) and is now properly documented. See https://docs.part-db.de/usage/keybindings.html

blue3121, 2023-04-25:
Thank you ! Great work !!!

commit bc65a18f (2023-04-23): Added greek characters to special characters plugin in CKEDITOR

Fixes #275