Part-DB/Part-DB-server #285: ClassNotFoundError for PurgerInterface when importing data from PartKeepr

mindsolve, 2023-04-30:
**Describe the bug**
When trying to import a PartKeepr database XML on a clean installation, the command to import the database fails with a `Symfony\Component\ErrorHandler\Error\ClassNotFoundError`.

**To Reproduce**
Steps to reproduce the behavior:

Follow the ["How to migrate"](https://docs.part-db.de/partkeepr_migration.html#how-to-migrate) steps exactly on a fresh Part-DB installation. Observe the error message on step 3.

**Expected behavior**
The parkeepr database to be imported without errors, as described in the documentation.

**Screenshots**
No screenshots, but here is the full command output:

```
root@ng-partdb:/var/www/partdb# sudo -u www-data php bin/console -vvv partdb:migrations:import-partkeepr /var/www/pk.xml

 ! [NOTE] This command is still in development. If you encounter any problems, please report them to the issue tracker
 !        on GitHub.

 [WARNING] This command will delete all existing data in the database (except users). Make sure that you have no
           important data in the database before you continue!

 Please type "DELETE ALL DATA" to continue. []:
 > DELETE ALL DATA

22:11:42 CRITICAL  [console] Error thrown while running command "-vvv partdb:migrations:import-partkeepr '/var/www/pk.xml'". Message: "Interface "Doctrine\Common\DataFixtures\Purger\PurgerInterface" not found"
[
  "exception" => Error {
    #message: "Interface "Doctrine\Common\DataFixtures\Purger\PurgerInterface" not found"
    #code: 0
    #file: "/var/www/partdb/src/Doctrine/Purger/ResetAutoIncrementORMPurger.php"
    #line: 48
    trace: {
      /var/www/partdb/src/Doctrine/Purger/ResetAutoIncrementORMPurger.php:48 {
        include^
        ›  */
        › class ResetAutoIncrementORMPurger implements PurgerInterface, ORMPurgerInterface
        › {
      }
      /var/www/partdb/vendor/composer/ClassLoader.php:582 { …}
      /var/www/partdb/vendor/composer/ClassLoader.php:433 { …}
      /var/www/partdb/src/Services/ImportExportSystem/PartKeeprImporter/PKImportHelper.php:47 { …}
      /var/www/partdb/src/Command/Migrations/ImportPartKeeprCommand.php:94 { …}
      /var/www/partdb/vendor/symfony/console/Command/Command.php:298 { …}
      /var/www/partdb/vendor/symfony/console/Application.php:1058 { …}
      /var/www/partdb/vendor/symfony/framework-bundle/Console/Application.php:96 { …}
      /var/www/partdb/vendor/symfony/console/Application.php:301 { …}
      /var/www/partdb/vendor/symfony/framework-bundle/Console/Application.php:82 { …}
      /var/www/partdb/vendor/symfony/console/Application.php:171 { …}
      /var/www/partdb/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:54 { …}
      /var/www/partdb/vendor/autoload_runtime.php:35 { …}
      /var/www/partdb/bin/console:11 { …}
    }
  },
  "command" => "-vvv partdb:migrations:import-partkeepr '/var/www/pk.xml'",
  "message" => "Interface "Doctrine\Common\DataFixtures\Purger\PurgerInterface" not found"
]
[
  "token" => null,
  "command" => [
    "name" => "partdb:migrations:import-partkeepr",
    "arguments" => [
      "command" => "partdb:migrations:import-partkeepr",
      "file" => "/var/www/pk.xml"
    ]
  ]
]
22:11:42 DEBUG     [console] Command "-vvv partdb:migrations:import-partkeepr '/var/www/pk.xml'" exited with code "1"
[
  "command" => "-vvv partdb:migrations:import-partkeepr '/var/www/pk.xml'",
  "code" => 1
]
[
  "token" => null,
  "command" => [
    "name" => "partdb:migrations:import-partkeepr",
    "arguments" => [
      "command" => "partdb:migrations:import-partkeepr",
      "file" => "/var/www/pk.xml"
    ]
  ]
]

In ResetAutoIncrementORMPurger.php line 48:

  [Symfony\Component\ErrorHandler\Error\ClassNotFoundError]
  Attempted to load interface "PurgerInterface" from namespace "Doctrine\Common\DataFixtures\Purger".
  Did you forget a "use" statement for another namespace?

Exception trace:
  at /var/www/partdb/src/Doctrine/Purger/ResetAutoIncrementORMPurger.php:48
 include() at /var/www/partdb/vendor/composer/ClassLoader.php:582
 Composer\Autoload\{closure}() at /var/www/partdb/vendor/composer/ClassLoader.php:433
 Composer\Autoload\ClassLoader->loadClass() at /var/www/partdb/src/Services/ImportExportSystem/PartKeeprImporter/PKImportHelper.php:47
 App\Services\ImportExportSystem\PartKeeprImporter\PKImportHelper->purgeDatabaseForImport() at /var/www/partdb/src/Command/Migrations/ImportPartKeeprCommand.php:94
 App\Command\Migrations\ImportPartKeeprCommand->execute() at /var/www/partdb/vendor/symfony/console/Command/Command.php:298
 Symfony\Component\Console\Command\Command->run() at /var/www/partdb/vendor/symfony/console/Application.php:1058
 Symfony\Component\Console\Application->doRunCommand() at /var/www/partdb/vendor/symfony/framework-bundle/Console/Application.php:96
 Symfony\Bundle\FrameworkBundle\Console\Application->doRunCommand() at /var/www/partdb/vendor/symfony/console/Application.php:301
 Symfony\Component\Console\Application->doRun() at /var/www/partdb/vendor/symfony/framework-bundle/Console/Application.php:82
 Symfony\Bundle\FrameworkBundle\Console\Application->doRun() at /var/www/partdb/vendor/symfony/console/Application.php:171
 Symfony\Component\Console\Application->run() at /var/www/partdb/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:54
 Symfony\Component\Runtime\Runner\Symfony\ConsoleApplicationRunner->run() at /var/www/partdb/vendor/autoload_runtime.php:35
 require_once() at /var/www/partdb/bin/console:11

partdb:migrations:import-partkeepr [--no-projects] [--import-users] [--] <file>
```

**Server Side**
- Part-DB Version: Part-DB master @ `v1.3.2` (`021e28aca877b27c4efc8f495f35575fe5df1176`)
- PHP Version: PHP `8.1.2-1ubuntu2.11`
- Database Server: none, sqlite
- Server OS: `Ubuntu 22.04.2 LTS`

**Desktop (please complete the following information):**
none, via console on server

**Smartphone (please complete the following information):**
none, via console on server

**Additional context**
Add any other context about the problem here.

jbtronics, 2023-04-30:
The class missing (which is used to purge the database before migration) is part of the doctrine datafixtures package. It seems that this only a dev requirement and therefore is not installed when running composer with `--no-dev`. I will fix that in the composer.json file.

For now, running `composer install -o` (without the `--no-dev`) in the Part-DB folder and then try the migration again, should do the trick.

mindsolve, 2023-04-30:
Thanks a lot! That worked, though I'm running into another problem, sooo you're going to have another issue shortly :D

jbtronics, 2023-05-01:
Fixed now

closed 2023-05-01 by jbtronics