Part-DB/Part-DB-server #287: DB migration issue

dirk1506, 2023-05-03:
Hi,
I have got problems with the migration of the existing part-db 0.5.15 (stable) version into the new one Part-DB server version 1.3.2. 
Although, the old legacy Parts-DB 0.5.15 is running just fine on my web hoster (all-inkl), with all my comprehensive electronic parts included in the DB, I just had the idea to test the new Part-DB server version as well.
The installation as above on "all-inkl" premium account runs on php 8.1 and MariaDB 10.5.19. 

So, I tried to test the Part-DB-Server on Ubuntu LXC on my internal Proxmox server first. Here my configuration:

Ubuntu 22.04 Standard LXC running on Proxmox 
Apache2, php 8.1, MariaDB 16.6.12, composer 2.5.5, node 18.16.0, yarn 1.22.19

For DB migration I installed the legacy Part-DB and imported a data base backup into it from my existing part-DB installation on my web hoster. It runs perfect.
But the migration via 
“php bin/console cache:clear” and “php bin/console doctrine:migrations:migrate” fails:
Here the print out:
[image]

I am wondering what happen. I am not a real data base expert, just the basics needed. I switched off as well during the DB import the option for foreign key checks but no success. Also checked google but have not a real glue.
I guess it has something to do with the different MariaDB versions. It would be a pity if I could not migrate my old data base into the new one. Without, it does not make sense to use the new one. 
I would appreciate any hints. Thanks a lot.

dirk1506, 2023-05-05:
OK. I changed to Debian 11 and installed PartDB-server as per the the installation guide, but now with MariaDB 10.5.19. The old Part-DB works perfect on my Proxmox server, however migration to the new one fails again with the same error as above in my first message. I imported the backups from my web hoster via phymyadmin several times, over and over onto an existing DB  and also over an empty one. The same result. I think I have to declare it as a bug. Sorry. I am a bit dissapointed. Because a smooth migration is one of the most importent feature for using the new partdb-server. I am fiddling around with that issue for almost two days and I am about to give it up. Although I like this tool and appriciate the good work of the developer. But it also seems to me, here is not that much activity. A pity... I proceed with the old one.

jbtronics, 2023-05-06:
The database schema of the old Part-DB versions was not very consistent, and there seems to be some subtle differences across different Part-DB installations. For my old database, the migration process works smooth, but it seems that you have some different foreign keys and indices structures.

To debug this, I need a SQL dump of your database (structure).

jbtronics, 2023-05-07:
I think I have found the issue. Please update Part-DB to the latest commit from the master branch (by switching with `git checkout master` and `git pull`) and retry the migration. I guess that should be fixed now.

dirk1506, 2023-05-08:
Thank you very much indeed. I confirm it is fixed now. Now it runs in my LXC container very well. Unfortunately, at the moment I am not in the position to install the new PartDB-Server on my webhoster "all-inkl", even I have a premium account. Something got wrong with the yast installation. But this is an issue of the web host installation. In the worst case I have to decide with which one I have to proceed. Let me test the new one at first and then I will decide. Thanks again for the good work.

closed 2023-05-08 by dirk1506