Part-DB/Part-DB-server #290: Remove all content when not logged in

ScaredyCat, 2023-05-18:
I could not find anything to disable the default functionality of displaying sidebar content before users log in.

**Is your feature request related to a problem? Please describe.**

No one, who is not logged in should be able to see any parameters or search data, let alone add new supplier or search for it. Before logging in none of the menus/ options ( Scanner, categories, projects, tools) should be available for use.

**Describe the solution you'd like**
Show a login screen only. Once logged in content should be shown.

**Additional context**
This seems to have a lot of potential for data leaks. I honestly can't fathom why any of these options are available at all to a user who is not logged in.

jbtronics, 2023-05-18:
You can configure what a non logged-in user can see/do, by configuring the permissions of the anonymous user. 

By default the anonymous user has read access, but you can set all permissions to forbid, then non logged in users can see nothing (just the banner on the homepage).

ScaredyCat, 2023-05-18:
Removing all permissions from the anonymous user does **not** remove all items from the sidebar.

virtadpt, 2023-05-18:
@ScaredyCat Can confirm.  I've been seeing this for a while (and tried stripping everything from the anonymous user).

jbtronics, 2023-05-22:
With the anonymous user set to "forbid all" you should see something like this: 

[image]

While the base structure with the configured number of sidebar panels is still visible, they have no content, and show no (sensitive) information. So I do not see a big problem with that behavior, or what do you expect?

whc2001, 2023-06-07:
IMHO something like PartKeepr login screen is very good. Only a login window with username and password, not even showing the welcome screen and sidebar.

jbtronics, 2023-08-20:
With the latest commit, the access to the homepage is forbidden, if a user has no permissions. This will be part of Part-DB 1.7.

So just apply the "Forbid all" permission preset to the anonymous user and just a login dialog will be shown when accessing Part-DB without login.

closed 2023-08-20 by jbtronics