Part-DB/Part-DB-server #297: Label PDF Empty: TypeError: can't access property "model", e is undefined

LovelyA72, 2023-06-04:
**Describe the bug**
When generating a part's label the label PDF is empty with nothing on it.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a part
2. Click on Label generator
3. Click "Update"
4. PDF is empty with error in the browser's console

**Screenshots**
[image]
[image]

**Server Side**
- Part-DB Version: Part-DB Docker (https://hub.docker.com/layers/jbtronics/part-db1/latest/images/sha256-3acf349d9081e7cf690964b035da2aac307ccf21c1a68cc9429a1f8640de7077?context=explore)
- PHP Version: Docker version built-in
- Database Server SQLite 

**Desktop (please complete the following information):**
 - OS: Windows 10 21H2
 - Browser Firefox 114.0b9 (also tested on Edge)

LovelyA72, 2023-06-04:
Update: This was an operation error. Contents will only appear when a bar code and contents are filled out. It will be nice if the system will alert the user that nothing will be generated.

jbtronics, 2023-06-27:
A warning is implemented in the upcoming 1.5 release

closed 2023-06-27 by jbtronics