Part-DB/Part-DB-server #298: Docker logs contains only apache logs

misaz, 2023-06-07:
I installed Part-DB using custom docker-compose script. While it is customized it is very similar to script for mysql at but similar to demo at https://docs.part-db.de/installation/installation_docker.html. For partdb image I use following env variables:

- DATABASE_URL=...
- APP_ENV=docker
- DEFAULT_LANG=...
- DEFAULT_TIMEZONE=...
- BASE_CURRENCY=...
- INSTANCE_NAME=...
- ALLOW_ATTACHMENT_DOWNLOADS=...

The docker logs command (and docker-compose logs also) produces only logs from apache.

```
AAA.BBB.CCC.DDD - - [07/Jun/2023:21:54:00 +0000] "POST /en/project/2 HTTP/1.1" 302 715 "XXXXXXXXXXXXXXXXX/en/project/2/edit" "YYYYYYYYYYUSERAGENTYYYYYYYYYY"
AAA.BBB.CCC.DDD - - [07/Jun/2023:21:54:00 +0000] "GET /en/project/new HTTP/1.1" 200 13290 "XXXXXXXXXXXXXXXXX/en/project/2/edit" "YYYYYYYYYYUSERAGENTYYYYYYYYYY"
AAA.BBB.CCC.DDD - - [07/Jun/2023:21:54:00 +0000] "GET /en/tree/categories HTTP/1.1" 200 9120 "XXXXXXXXXXXXXXXXX/en/project/new" "YYYYYYYYYYUSERAGENTYYYYYYYYYY"
AAA.BBB.CCC.DDD - - [07/Jun/2023:21:54:00 +0000] "GET /en/tree/devices HTTP/1.1" 200 1072 "XXXXXXXXXXXXXXXXX/en/project/new" "YYYYYYYYYYUSERAGENTYYYYYYYYYY"
AAA.BBB.CCC.DDD - - [07/Jun/2023:21:54:00 +0000] "GET /en/tree/tools HTTP/1.1" 200 3665 "XXXXXXXXXXXXXXXXX/en/project/new" "YYYYYYYYYYUSERAGENTYYYYYYYYYY"
AAA.BBB.CCC.DDD - - [07/Jun/2023:21:54:03 +0000] "GET /en/project/1/edit HTTP/1.1" 200 21834 "XXXXXXXXXXXXXXXXX/en/project/new" "YYYYYYYYYYUSERAGENTYYYYYYYYYY"
AAA.BBB.CCC.DDD - - [07/Jun/2023:21:54:04 +0000] "POST /en/project/1/edit HTTP/1.1" 200 18750 "XXXXXXXXXXXXXXXXX/en/project/1/edit" "YYYYYYYYYYUSERAGENTYYYYYYYYYY"
```

I tried to enter bash shell in the docker container but the only file in var/log is docker_deprecations.log with entries not related to http requests.

```
root@8e1057a79de7:/var/www/html# ls -l var/log/
total 40
-rw-r--r-- 1 www-data www-data 39800 Jun  6 10:35 docker_deprecations.log
```

How can I get partdb logs with more details about errors when using docker?

jbtronics, 2023-06-10:
There was an configuration missing, which prevented Part-DB logs to show up in docker logs.
This is now fixed and should work properly when using the docker image of Part-DB 1.4.1

closed 2023-06-10 by jbtronics

misaz, 2023-06-10:
I think you meant 1.4.2, did not you? 1.4.1 was relased 4 days ago, but comot is 0 days ago.

jbtronics, 2023-06-10:
Yeah I meant 1.4.2
That was just a typo

misaz, 2023-06-10:
Thanks. Have a nice day.