Part-DB/Part-DB-server #299: Importing PartKeepr dumped XML kills all user in the database

whc2001, 2023-06-08:
**Describe the bug**
Following the documentation, after running the PartKeepr import (without importing PartKeepr users) and clear cache command, all local users created from the installation (including admin) got deleted from the database. The WebUI is unable to load after this. The documentation said it should keep the local users.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the [installation tutorial](https://docs.part-db.de/installation/installation_guide-debian.html), until the WebUI is available and can be logged in with admin account.
2. Follow the [PartKeepr import tutorial](https://docs.part-db.de/partkeepr_migration.html) to import the PartKeepr XML dump and clear the cache.
3. After this, the WebUI only shows 500 Error when opened
4. After inspecting the database, there is nothing in the `user` table.

**Expected behavior**
The `admin` user should at least still exist in the database.

**Screenshots**

Before importing:
[image]

After importing:
[image]

**Server Side**
- Part-DB Version: 1.4.1 (master/)
- PHP Version: 8.1.18 (SAPI: apache2handler)
- Database Server: 10.5.19-MariaDB-0+deb11u2

**Desktop (please complete the following information):**
[Not Related]

**Smartphone (please complete the following information):**
[Not Related]

**Additional context**
N/A

whc2001, 2023-06-08:
I think I know why after inserting some debug print into the database purger logic.

[image]

At here:

https://github.com/Part-DB/Part-DB-server/blob/636776c5313338382f36f45823906c1ee26d9b48/src/Services/ImportExportSystem/PartKeeprImporter/PKImportHelper.php#L46-L47

Not only quotes are required, but also backticks for some reason. I did this dirty hack and it's solved:

```php
  // If the table is excluded, skip it as well
  $tbl = trim($tbl, "\"");
  $tbl = trim($tbl, "`");
  if (in_array($tbl, $this->excluded, true)) {
```

```php
  // Quotes are already handled in the Purger
  $purger = new ResetAutoIncrementORMPurger($this->em, ['users', 'groups', 'u2f_keys', 'internal', 'migration_versions']);
  $purger->purge();

```

jbtronics, 2023-06-10:
Now fixed in Part-DB 1.4.2

closed 2023-06-10 by jbtronics