Part-DB/Part-DB-server #304: Database Migration from 0.5.15 fails with  " Unknown column 'perms_labels' in 'groups' "

Phiwatec, 2023-06-13:
When running `sudo -u www-data php bin/console doctrine:migrations:migrate` as in the migration guide the migration fails with the following error:
[image]

I first upgraded to PartDB Version 0.5.15 as recommended in the Upgrade Guide. 
Then I followed the steps of which all except the mentioned step worked. 

Rerunning the migration on same Database results in a different error, which is okay as it is missing already migrated columns:
[image]

**Server Side**
- Part-DB Version: 1.4.2 / 0.5.15-legacy  (Different folders) 
- PHP Version: 8.2
- Database Server: MariaDB 10.11.3
- OS: Debian 12

jbtronics, 2023-06-16:
Thats strange. Is your database up to date in the legacy Part-DB? This column should have been created during older database migration in Part-DB-legacy.
I assume that this is some issue with your database (or some wrong migration in the past) and not really a migration issue of Part-DB 1.x

However your problem should be fixable, by reverting your database to the backup and running the following SQL command on the database (using phpmyadmin or command line):

```
ALTER TABLE `groups` ADD `perms_labels` SMALLINT  DEFAULT 0  NOT NULL AFTER `perms_tools`
```

Phiwatec, 2023-06-18:
That fixed it. Thank you very much.
Greetings

closed 2023-06-18 by Phiwatec