Part-DB/Part-DB-server #309: Display all parts

blue3121, 2023-07-01:
Steps to reproduce the behavior:

1) View parts in a category with more than 50 parts
2) Select "All" to view them all
3) Reselect the category to redisplay them
Now, with "All" selected, only 50 parts are displayed

--------

- Part-DB Version: 1.5.0-dev
- PHP Version: 8.1.2
- Database Server: sqlite

 - OS: Kubuntu 22.04 LTS
 - Browser: Firefox
 - Version: 114.0.1

jbtronics, 2023-07-02:
This is now fixed (the page length selector now correctly show 50 items on reload of a table)

closed 2023-07-02 by jbtronics

blue3121, 2023-07-04:
It is preferable not to reset the number of visible parts but to display all the parts based on the selection otherwise you have to update this number every time.
I do this operation every time I have to insert many parts in the same category, checking the correct insertion between one insertion and another.