Part-DB/Part-DB-server #314: Label scaling

vidarhag, 2023-07-05:
I have been trying to print a partnumber label including a QR code to use for a Dymo 550, 19x51mm label. It seems like the QR code get cropped. Is it possible to scale down the QR part in the label to fit for smaller labels ? Or, maybe export to other file formats ?

jbtronics, 2023-07-05:
That seems to be caused by a combination of the page margin and qr code positioning, I will try to fix that for all. 
You can fix that by adding 
```css
.qr {
    max-width: 80%;
    max-height: 100%;
}
```
in the "additional styles (CSS) field" in the advanced tab.

For these small labels the page margins are quite big. You can control them by putting
````css
*@page {
    margin: 4px 4px;
}
```
in the styles field and changing them for your needs.

jbtronics, 2023-07-08:
The height fix is applied to all users in Part-DB 1.5.2

closed 2023-07-08 by jbtronics