Part-DB/Part-DB-server #315: Category Parameters not applied to items in category

oXL0C0Xo, 2023-07-06:
**Describe the bug
Can add parameters to a category, but parameter list is not applied to products placed in that category.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a parameter to a category
2. Add a part to that category
3. Edit part, and look at parameters tab

**Expected behavior**
I'm just assuming here, but if you add a parameter to a category, it should display that parameter on the products in that category.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Server Side**
- Part-DB Version: [e.g. Part-DB 1.0 (for dev versions commit hash is useful]
- PHP Version: [e.g. PHP 7.4.3]
- Database Server [e.g. MySQL 5.7]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Firefox
 - Version 114.0.2

**Smartphone (please complete the following information):**
N/A

**Additional context**
Add any other context about the problem here.

jbtronics, 2023-07-06:
No parameters on data structures (like categories, footprints, etc). does not work that way.
These parameters give information about the data structures itself (like think of the pin count or dimensions of a footprint, or some additional information about suppliers),, not of the parts contained inside them.
That is basically just a possibility to add additional information to a data structures in a structured way.

oXL0C0Xo, 2023-07-06:
So for instance if I am listing resistors, I have to create a "Resistance" parameter on each entry so that I can add that attribute to the item?

jbtronics, 2023-07-06:
Parameters are defined per Part.
If you want to create a lot of parts, with only small differences in one or two fields, you can clone the part from either the info page or use the dropdown on the part edit page "Save and clone".
You will then get a creation mask filled with all the infos from the original part, which you can change to the new needs

closed 2023-07-17 by oXL0C0Xo