Part-DB/Part-DB-server #316: Yarn build translator.js problem on Ubuntu 22.04

SilentResonance, 2023-07-08:
Hello,

first of all, thanks for this great electronics inventory tool :)
I am trying to install part-db from master (1.5.1).
After that, I would like to start developing/enhancing part-db.
The bug occurs in step [Install dependencies for Part-DB and build frontend](https://docs.part-db.de/installation/installation_guide-debian.html#install-dependencies-for-part-db-and-build-frontend) when running `sudo yarn build`

**To Reproduce**
Steps to reproduce the behavior:
1. On Ubuntu 22.04.
2. Follow the instructions [here](https://docs.part-db.de/installation/installation_guide-debian.html)
3. Error:
` ERROR  Failed to compile with 2 errors`
`Module build failed: Module not found:`
`"./assets/translator.js" contains a reference to the file "../var/translations/configuration".`
`This file can not be found, please check it for typos or update it if the file got moved.`
`"./assets/translator.js" contains a reference to the file "../var/translations".`
`This file can not be found, please check it for typos or update it if the file got moved.`

**Server Side**
- Part-DB Version: v1.5.1 (commit ddbf8b772583b061e3d77294cf47936f929cbe38)
- PHP Version: 8.1.2-1ubuntu2.13
- Database Server not yet installed/configured (is a later step)

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
 - Browser not used yet

**Repair attempts**
- Tried running all `sudo` command in documentation with `-u www-data`
- Changed import directory in `./assets/translator.js` from `../var/translations/configuration` to `./translations`, but in the repository there is no configuration in the translations folder.

Did I miss an important step? Is there a step missing in the documentation? Or is there a required file not pushed in the repository?

Thanks in advance.

P.S.: I am completely new to symfony development and all the tooling (composer, yarn) that comes with it. Any hint on how to fix this and get a production version running is appreciated.

SilentResonance, 2023-07-08:
Found my error....

using the Ubuntu delivered composer version (2.2.x) did not work. Also the yarn version is too old.
Had to `sudo apt install php8.1-intl` manually although I copied the command further up, which contains exactly that command.
And `sudo apt install php-mysql` is missing in the doc for the mysql part as a prerequisite.

Works with the latest version of composer and yarn 18.x (as described in documentation)

closed 2023-07-08 by SilentResonance

jbtronics, 2023-07-08:
> using the Ubuntu delivered composer version (2.2.x) did not work

that version is pretty new.

> 3\. Error:
>     ` ERROR  Failed to compile with 2 errors`
>     `Module build failed: Module not found:`
>     `"./assets/translator.js" contains a reference to the file [...]

The files, which were missing here, are just generated on a cache warmup of Part-DB. This should happen automatically after installing the composer dependencies.  If that not happens, you can do a `php bin/console cache:clear` to trigger it manually.