Part-DB/Part-DB-server #318: Uncaught PHP Exception when redirect to SAML SSO

cubesky, 2023-07-10:
**Describe the bug**
When using SAML login method, redirect to SSO server is denied.

When I manully look at PartDB logs for the target redirect url, I copy it and paste to my browser (Keycloak url), login and redirect back, PartDB successfully logined. 

**To Reproduce**
1. Config the SAML following the Configuration document with keyclock server. (Keycloak on `https://sso.server.example.com/`)
2. Config PartDB when docker.
3. Press the SSO button.

**Expected behavior**
Redirect to the Keycloak login page.

**Screenshots**
[image]

**Server Side**
- Part-DB Version: Part-DB 1.5.2
- PHP Version: PHP 8.1.20
- Database Server: Sqlite

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Chrome
 - Version: 114.0.5735.199

**Smartphone (please complete the following information):**
 - Device: Not Use
 - OS: Not Use
 - Browser Not Use
 - Version Not Use

**Additional context**
The following log replaced my domain with `example.com`

```
partdb    | [10-Jul-2023 20:10:16] NOTICE: fpm is running, pid 24
partdb    | [10-Jul-2023 20:10:16] NOTICE: ready to handle connections
partdb    | [10-Jul-2023 20:10:16] NOTICE: systemd monitor interval set to 10000ms
partdb    | AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 192.168.64.2. Set the 'ServerName' directive globally to suppress this message
partdb    | AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 192.168.64.2. Set the 'ServerName' directive globally to suppress this message
partdb    | [Mon Jul 10 20:10:16.737909 2023] [mpm_event:notice] [pid 1:tid 140665374002496] AH00489: Apache/2.4.56 (Debian) configured -- resuming normal operations
partdb    | [Mon Jul 10 20:10:16.738109 2023] [core:notice] [pid 1:tid 140665374002496] AH00094: Command line: 'apache2 -D FOREGROUND'
partdb    | - -  10/Jul/2023:20:10:21 +0000 "GET /index.php" 302
partdb    | 10.42.1.2 - - [10/Jul/2023:20:10:21 +0000] "GET / HTTP/1.1" 302 875 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36"
partdb    | - -  10/Jul/2023:20:10:24 +0000 "GET /index.php" 200
partdb    | 10.42.1.2 - - [10/Jul/2023:20:10:24 +0000] "GET /en/ HTTP/1.1" 200 6249 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36"
partdb    | - -  10/Jul/2023:20:10:25 +0000 "POST /index.php" 204
partdb    | 10.42.1.2 - - [10/Jul/2023:20:10:25 +0000] "POST /csp/report HTTP/1.1" 204 748 "https://partdb.example.com/en/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36"
partdb    | - -  10/Jul/2023:20:10:25 +0000 "GET /index.php" 200
partdb    | 10.42.1.2 - - [10/Jul/2023:20:10:25 +0000] "GET /en/tree/devices HTTP/1.1" 200 937 "https://partdb.example.com/en/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36"
partdb    | - -  10/Jul/2023:20:10:25 +0000 "GET /index.php" 200
partdb    | 10.42.1.2 - - [10/Jul/2023:20:10:25 +0000] "GET /en/tree/tools HTTP/1.1" 200 2943 "https://partdb.example.com/en/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36"
partdb    | - -  10/Jul/2023:20:10:25 +0000 "GET /index.php" 200
partdb    | 10.42.1.2 - - [10/Jul/2023:20:10:25 +0000] "GET /en/tree/categories HTTP/1.1" 200 936 "https://partdb.example.com/en/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36"
partdb    | - -  10/Jul/2023:20:10:27 +0000 "GET /index.php" 200
partdb    | 10.42.1.2 - - [10/Jul/2023:20:10:27 +0000] "GET /en/login?_target_path=/ HTTP/1.1" 200 6226 "https://partdb.example.com/en/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36"
partdb    | - -  10/Jul/2023:20:10:27 +0000 "POST /index.php" 204
partdb    | 10.42.1.2 - - [10/Jul/2023:20:10:27 +0000] "POST /csp/report HTTP/1.1" 204 748 "https://example.com/en/login?_target_path=/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36"
partdb    | {"message":"Matched route \"saml_login\".","context":{"route":"saml_login","route_parameters":{"_route":"saml_login","_controller":"Nbgrp\\OneloginSamlBundle\\Controller\\Login","idp":null},"request_uri":"https://partdb.example.com/saml/login","method":"GET"},"level":200,"level_name":"INFO","channel":"request","datetime":"2023-07-10T20:10:28.250504+00:00","extra":{"token":null,"url":"/saml/login","ip":"10.42.10.4","http_method":"GET","server":"partdb.example.com","referrer":"https://partdb.example.com/en/login?_target_path=/"}}
partdb    | {"message":"Checking for authenticator support.","context":{"firewall_name":"main","authenticators":4},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-07-10T20:10:28.251811+00:00","extra":{"token":null,"url":"/saml/login","ip":"10.42.10.4","http_method":"GET","server":"partdb.example.com","referrer":"https://partdb.example.com/en/login?_target_path=/"}}
partdb    | {"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Scheb\\TwoFactorBundle\\Security\\Http\\Authenticator\\TwoFactorAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-07-10T20:10:28.251834+00:00","extra":{"token":null,"url":"/saml/login","ip":"10.42.10.4","http_method":"GET","server":"partdb.example.com","referrer":"https://partdb.example.com/en/login?_target_path=/"}}
partdb    | {"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Scheb\\TwoFactorBundle\\Security\\Http\\Authenticator\\TwoFactorAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-07-10T20:10:28.251855+00:00","extra":{"token":null,"url":"/saml/login","ip":"10.42.10.4","http_method":"GET","server":"partdb.example.com","referrer":"https://partdb.example.com/en/login?_target_path=/"}}
partdb    | {"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Nbgrp\\OneloginSamlBundle\\Security\\Http\\Authenticator\\SamlAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-07-10T20:10:28.251868+00:00","extra":{"token":null,"url":"/saml/login","ip":"10.42.10.4","http_method":"GET","server":"partdb.example.com","referrer":"https://partdb.example.com/en/login?_target_path=/"}}
partdb    | {"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Nbgrp\\OneloginSamlBundle\\Security\\Http\\Authenticator\\SamlAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-07-10T20:10:28.251882+00:00","extra":{"token":null,"url":"/saml/login","ip":"10.42.10.4","http_method":"GET","server":"partdb.example.com","referrer":"https://partdb.example.com/en/login?_target_path=/"}}
partdb    | {"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\FormLoginAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-07-10T20:10:28.251895+00:00","extra":{"token":null,"url":"/saml/login","ip":"10.42.10.4","http_method":"GET","server":"partdb.example.com","referrer":"https://partdb.example.com/en/login?_target_path=/"}}
partdb    | {"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\FormLoginAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-07-10T20:10:28.251909+00:00","extra":{"token":null,"url":"/saml/login","ip":"10.42.10.4","http_method":"GET","server":"partdb.example.com","referrer":"https://partdb.example.com/en/login?_target_path=/"}}
partdb    | {"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\RememberMeAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-07-10T20:10:28.251922+00:00","extra":{"token":null,"url":"/saml/login","ip":"10.42.10.4","http_method":"GET","server":"partdb.example.com","referrer":"https://partdb.example.com/en/login?_target_path=/"}}
[... cut ...]
        proxy_http_version 1.1;
        proxy_redirect off;
        proxy_set_header REMOTE-HOST $remote_addr;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection "Upgrade";
        proxy_set_header Host $host;
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header X-Scheme $scheme;
        proxy_set_header X-Forwarded-Proto $scheme;
        proxy_pass http://127.0.0.1:18580;
    }
}
```

Keycloak

[image]

jbtronics, 2023-07-10:
Interesting. That is not really an exception caused by some erroneous code, but is a basically a security feature, which should prevent url redirection attack, as a last security net. However, as SAML requires an external redirect, it prevents SAML in default configuration. (In my dev environment, keycloak and Part-DB are both on localhost, so I never noticed it before). 

I will try to find a solution, that the domain is automatically whitelisted. But for now you can go to `config/packages/nelmio_security.yaml.` andf find the line with `external_redirects`. Change the section to the following:

```yaml
    external_redirects:
        abort: true
        log: true
        allow_list:
             - your.idp.domain.com
```

Or you can disable the external redirect blocking completly by setting the `abort` key to `false`.
Afterward you have to run `php bin/console cache:clear` and the redirect should work as intended.

cubesky, 2023-07-11:
Ok! It works!

jbtronics, 2023-07-11:
This will be fixed in version 1.6.0

closed 2023-07-11 by jbtronics