Part-DB/Part-DB-server #321: Yarn threads failing with code 2

bobbienet, 2023-07-17:
Description:
Yarn appears to be having issues installing properly, not sure if i have the wrong yarn or if it's misconfigured on my end,

- Part-DB Version: 1.5.2
- PHP Version: 8.1.2-1ubuntu2.13 (cli)
- Database Server: 15.1 Distrib 10.6.12-MariaDB
- 
 - OS: ubuntu 22.04
 - Browser Chrome

**To Reproduce**
1. installed yarn via sudo apt install --no-install-recommends yarn
2. yarn version 1.22.19
3. sudo yarn build outputs:
partdb:/var/www/partdb$ sudo yarn build
```
yarn run v1.22.19
$ encore production --progress
Running webpack ...

[webpack-cli] Failed to load '/var/www/partdb/webpack.config.js' config
[webpack-cli] /var/www/partdb/node_modules/css-minimizer-webpack-plugin/node_modules/jest-worker/build/index.js:126
      enableWorkerThreads: this._options.enableWorkerThreads ?? false,
                                                              ^

SyntaxError: Unexpected token '?'
    at wrapSafe (internal/modules/cjs/loader.js:915:16)
    at Module._compile (internal/modules/cjs/loader.js:963:27)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/var/www/partdb/node_modules/css-minimizer-webpack-plugin/dist/index.js:17:5)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
error Command failed with exit code 2.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command
```

**Additional context**
got a 500 error page, ran yarn install and yarn build, then php bin/console cache:clear and that has caused the debug output to dissapear. following other bugs i looked at var/log/prod.log and I see lots of unhandled and uncaught exceptions with the following:

`
Cannot rename \"/tmp/url_matching_routes.phpfAc4U6\" to \"/var/www/partdb/var/cache/prod/url_matching_routes.php\": rename(/tmp/url_matching_routes.phpfAc4U6,/var/www/partdb/var/cache/prod/url_matching_routes.php): Permission denied","context"
`

jbtronics, 2023-07-17:
What nodejs version are you running? You can check that with `node --version`. Part-DB requires at least nodejs 18.

The error message looks like they are caused by some wrong file permissions. Have you chowned all files in the Part-DB folder to the apache user and have given write access to all folders?

bobbienet, 2023-07-17:
Mislabeled as bug and not issue; apologies.

it appears to be working now that i've purged node via apt and installed with 
`sudo npm cache clean -f
sudo npm install -g n
sudo n stable
 sudo apt-get install --reinstall nodejs-legacy 
`
yarn install and yarn build need to be run as the apache user .

got a few warning during yarn install regarding 
`warning Skipping preferred cache folder "/var/www/.cache/yarn" because it is not writable.
warning Selected the next writable cache folder in the list, will be "/tmp/.yarn-cache-33".
warning Cannot find a suitable global folder. Tried these: "/usr/local, /var/www/.yarn"
`
However I'm able to login to the web UI now and access the add part options. 

Much appreciated!

closed 2023-07-17 by bobbienet