Part-DB/Part-DB-server #326: Digikey API

pdo59, 2023-07-20:
When I try to click on "Connect OAuth" and after logging in to the Digikey site I always get this message:
{"ErrorCode" : "invalid_request", "Error" :"Invalid redirection uri http://partdb.lan/en/oauth/client/ip_digikey_oauth/check"}

- Part-DB Version: 1.6.0
- PHP Version: 8.1.21
- Database Server: 10.5.19-MariaDB-0+deb11u2

Desktop:
 - OS: Debian or MacOs
 - Browser: Firefox 102.13.0esr or Safari 16.5.2

jbtronics, 2023-07-20:
Is this the response digikey give or Part-DB?
(So is page that shows this issue on your Part-DB domain or a digikey one?)

pdo59, 2023-07-20:
Is the Digikey site: https://api.digikey.com/v1/oauth2/redirect?code=UMqjJsQFP7JB-mKDhZHIHGehn-eYlEX70AoAAAAB&state=90fdf6f7-836e-4811-febd-faddb8105eb9

jbtronics, 2023-07-20:
I think you have to set your part-Domain (or maybe even the full path) as redirect URI in the digikey settings.

I thought this field does not matter, but I guess it defaults to localhost if not set (that's why it worked in my tests). If your part-db is reached via another domain you have to set it to another value.

pdo59, 2023-07-20:
My partDB system is in a virtual machine with Debian11. I configured apache2 to respond to http://partdb.lan

jbtronics, 2023-07-20:
> Eg. in digikey site there is a field "OAuth Callback" that I leaved empty

I think that is the parameter you have to change and provide the domain name of your partdb instance there

pdo59, 2023-07-20:
The parameter requires an "https://" domain, I tried to enter my local domain with https:// prefix but the result is the same: "Invalid redirection uri http://partdb.lan/en/oauth/client/ip_digikey_oauth/check".
However, as can be seen, the error does not contain https while the parameter entered in Digikey contains it; perhaps it is not the domain that must be entered, I believe that Digikey's response is returned to the starting url anyway, perhaps identify in partDb which is the page for the oauth check

jbtronics, 2023-07-20:
Then it is caused by the fact that your Part-DB instance is not reachable via HTTPS. After authorization the digikey Oauth provider redirects the browser back to a special Part-DB page, and Part-DB extract the authorization information from the query.

Digikey seems to enforce that the redirect URI is a https domain, so the authorization information is transferred encrypted to the application (which is reasonable, otherwise you can easily steal the API tokens during transfer).

So basically to use the OAuth authorization your Part-DB needs to be accessible via HTTPs (and you have to visit it via HTTPs connection, when you click on the button, so that the correct redirect is generated). Using a self-signed certificate should be enough.

pdo59, 2023-07-20:
Ok, I enable the https and let you know if all is ok. 
However at the moment I am not able to run partDb, after changing a category the system keeps giving me this message:
  "The server returned a "500 Internal Server Error".
Something is broken. Please let us know what you were doing when this error occurred. We will fix it as soon as possible. Sorry for any inconvenience caused. "
Can you help me?

jbtronics, 2023-07-20:
what do the logs of Part-DB say?

pdo59, 2023-07-20:
This is prod_deprecations.log: 
`[2023-07-20T16:25:57.584099+02:00] deprecation.INFO: User Deprecated: Accessing Doctrine\Common\Lexer\Token properties via ArrayAccess is deprecated, use the value, type or position property instead (Token.php:108 called by DocParser.php:691, https://github.com/doctrine/lexer/pull/79, package doctrine/lexer) {"exception":"[object] (ErrorException(code: 0): User Deprecated: Accessing Doctrine\\Common\\Lexer\\Token properties via ArrayAccess is deprecated, use the value, type or position property instead (Token.php:108 called by DocParser.php:691, https://github.com/doctrine/lexer/pull/79, package doctrine/lexer) at /var/www/partdb/vendor/doctrine/deprecations/lib/Doctrine/Deprecations/Deprecation.php:209)"} {"token":null,"url":"/","ip":"10.211.55.16","http_method":"GET","server":"partdb.lan","referrer":null}
`
And this is the prod.log, both contents are since I refresh the page
[prod.log](https://github.com/Part-DB/Part-DB-server/files/12110383/prod.log)

jbtronics, 2023-07-20:
The permissions of your Part-DB installation are messed up. Remove the var/cache folder in Part-DB and do a `chown apache:apache -R *` afterwards in the Part-DB folder (or however your webserver user is named)

pdo59, 2023-07-20:
...  the system seemed to work, but if I try to edit the categories it fails again with 500... the other menu items seem to work except the Categories and New part

pdo59, 2023-07-20:
> The permissions of your Part-DB installation are messed up. Remove the var/cache folder in Part-DB and do a `chown apache:apache -R *` afterwards in the Part-DB folder (or however your webserver user is named)

The problem still persist

pdo59, 2023-07-20:
There is a problem with inserting Attachments to categories: via SQL I set the "id_preview_attachment" field to NULL and now the system seems to work.

pdo59, 2023-07-20:
I enabled HTTPs in my site, now partDb works on https://partdb.lan, everything ok except the Digikey click "Connect OAuth" which gives the same error, but with "https" instead of "http":
`{"ErrorCode" : "invalid_request", "Error" :"Invalid redirection uri https://partdb.lan/en/oauth/client/ip_digikey_oauth/check"}`

pdo59, 2023-07-21:
Digikey API still doesn't work, I put the tokens (taken from another software where the API works) into the `OAuthCredentials` table, but the provider is still disabled. In the table I set the 'id' field to 1 and the 'name' field to 'digikey',

pdo59, 2023-07-21:
I tested. also a sandbox app in the Digikey site, the changed Client ID and Client Secret, but still not working, there is the response of the Digikey site: 
`{"ErrorResponseVersion":"3.0.0.0","StatusCode": 401,"ErrorMessage":"Invalid clientId","ErrorDetails":"clientId invalid for requested resource","RequestId":"40f800c2-a6d4-4047-bcb8-5a9df58ca577","ValidationErrors":[]}`

jbtronics, 2023-07-21:
> I tested. also a sandbox app in the Digikey site, the changed Client ID and Client Secret, but still not working, there is the response of the Digikey site: [...]

That is expected behavior. The sandbox have a completly different API endpoint, and these sandbox credentials are not recognized at the production endpoint

jbtronics, 2023-07-21:
> Digikey API still doesn't work, I put the tokens (taken from another software where the API works) into the `OAuthCredentials` table, but the provider is still disabled. In the table I set the 'id' field to 1 and the 'name' field to 'digikey',

The correct name for the token entry is `ip_digikey_oauth`. Also, you need to set an expiration date.

I would suspect, that digikey maybe do not like .lan domains (as its just a private TLD)

jbtronics, 2023-07-21:
I tried to mock your redirect URI, and digikey generates a redirect to the partdb.lan domain without any problems
[image]

Can you send an screenshot of the digikey configuration page?

jbtronics, 2023-07-21:
It works for me if the callback URI option on the digikey page is either empty or `https://partdb.lan/en/oauth/client/ip_digikey_oauth/check`

pdo59, 2023-07-21:
Sorry Jan,
"mea culpa". 
I used my client_id and secret from my other software while on the digikey developer site I had created a new app for partDB which generates new client_id and secret. 
This issue can be closed,
thank you for you support.

ewa, 2023-09-05:
I'm having a parallel issue, despite having confirmed that I'm using the right credentials for DigiKey.  I got it to where DigiKey itself seems to be happy with my OAuth request, but then Part-DB has some kind of negative reaction to the post-authorization-redirect request it gets from my browser.

[image]

The logs from that request are as follows:
`
2023-09-05 14:21:44 172.18.0.1 - - [05/Sep/2023:20:21:44 +0000] "GET /en/oauth/client/ip_digikey_oauth/connect HTTP/1.1" 302 2512 "https://localhost/en/tools/info_providers/providers" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/115.0.0.0 Safari/537.36"
2023-09-05 14:21:44 - -  05/Sep/2023:20:21:44 +0000 "GET /index.php" 302
2023-09-05 14:21:45 {"message":"Matched route \"oauth_client_check\".","context":{"route":"oauth_client_check","route_parameters":{"_route":"oauth_client_check","_controller":"App\\Controller\\OAuthClientController::check","_locale":"en","name":"ip_digikey_oauth"},"request_uri":"https://localhost/en/oauth/client/ip_digikey_oauth/check?code=XGLNj7ia&scope=&state=2dc934dd23fb59c04f6a5ec8537238a6","method":"GET"},"level":200,"level_name":"INFO","channel":"request","datetime":"2023-09-05T20:21:45.436845+00:00","extra":{"token":null,"url":"/en/oauth/client/ip_digikey_oauth/check?code=XGLNj7ia&scope=&state=2dc934dd23fb59c04f6a5ec8537238a6","ip":"172.18.0.1","http_method":"GET","server":"localhost","referrer":"https://localhost/"}}
2023-09-05 14:21:45 {"message":"Checking for authenticator support.","context":{"firewall_name":"main","authenticators":4},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-09-05T20:21:45.438229+00:00","extra":{"token":null,"url":"/en/oauth/client/ip_digikey_oauth/check?code=XGLNj7ia&scope=&state=2dc934dd23fb59c04f6a5ec8537238a6","ip":"172.18.0.1","http_method":"GET","server":"localhost","referrer":"https://localhost/"}}
2023-09-05 14:21:45 {"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Scheb\\TwoFactorBundle\\Security\\Http\\Authenticator\\TwoFactorAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-09-05T20:21:45.438248+00:00","extra":{"token":null,"url":"/en/oauth/client/ip_digikey_oauth/check?code=XGLNj7ia&scope=&state=2dc934dd23fb59c04f6a5ec8537238a6","ip":"172.18.0.1","http_method":"GET","server":"localhost","referrer":"https://localhost/"}}
2023-09-05 14:21:45 {"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Scheb\\TwoFactorBundle\\Security\\Http\\Authenticator\\TwoFactorAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-09-05T20:21:45.438267+00:00","extra":{"token":null,"url":"/en/oauth/client/ip_digikey_oauth/check?code=XGLNj7ia&scope=&state=2dc934dd23fb59c04f6a5ec8537238a6","ip":"172.18.0.1","http_method":"GET","server":"localhost","referrer":"https://localhost/"}}
2023-09-05 14:21:45 {"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Nbgrp\\OneloginSamlBundle\\Security\\Http\\Authenticator\\SamlAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-09-05T20:21:45.438275+00:00","extra":{"token":null,"url":"/en/oauth/client/ip_digikey_oauth/check?code=XGLNj7ia&scope=&state=2dc934dd23fb59c04f6a5ec8537238a6","ip":"172.18.0.1","http_method":"GET","server":"localhost","referrer":"https://localhost/"}}
2023-09-05 14:21:45 {"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Nbgrp\\OneloginSamlBundle\\Security\\Http\\Authenticator\\SamlAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-09-05T20:21:45.438288+00:00","extra":{"token":null,"url":"/en/oauth/client/ip_digikey_oauth/check?code=XGLNj7ia&scope=&state=2dc934dd23fb59c04f6a5ec8537238a6","ip":"172.18.0.1","http_method":"GET","server":"localhost","referrer":"https://localhost/"}}
2023-09-05 14:21:45 {"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\FormLoginAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-09-05T20:21:45.438295+00:00","extra":{"token":null,"url":"/en/oauth/client/ip_digikey_oauth/check?code=XGLNj7ia&scope=&state=2dc934dd23fb59c04f6a5ec8537238a6","ip":"172.18.0.1","http_method":"GET","server":"localhost","referrer":"https://localhost/"}}
2023-09-05 14:21:45 {"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\FormLoginAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-09-05T20:21:45.438310+00:00","extra":{"token":null,"url":"/en/oauth/client/ip_digikey_oauth/check?code=XGLNj7ia&scope=&state=2dc934dd23fb59c04f6a5ec8537238a6","ip":"172.18.0.1","http_method":"GET","server":"localhost","referrer":"https://localhost/"}}
2023-09-05 14:21:45 {"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\RememberMeAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-09-05T20:21:45.438316+00:00","extra":{"token":null,"url":"/en/oauth/client/ip_digikey_oauth/check?code=XGLNj7ia&scope=&state=2dc934dd23fb59c04f6a5ec8537238a6","ip":"172.18.0. [... cut ...]  OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/115.0.0.0 Safari/537.36"
2023-09-05 14:21:45 - -  05/Sep/2023:20:21:45 +0000 "GET /index.php" 500
2023-09-05 14:21:46 172.18.0.1 - - [05/Sep/2023:20:21:46 +0000] "GET /favicon.ico HTTP/1.1" 200 17353 "https://localhost/en/oauth/client/ip_digikey_oauth/check?code=XGLNj7ia&scope=&state=2dc934dd23fb59c04f6a5ec8537238a6" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/115.0.0.0 Safari/537.36"`

ewa, 2023-09-08:
Does DigiKey OAuth work for you?  Is this just my setup?

jbtronics, 2023-09-09:
> Does DigiKey OAuth work for you? Is this just my setup?

Just tried it again, and for me it just works fine. And your error trace basically says, that the response returned by Digikey don't contain the required fields, after Part-DB requests a token using the access code returned by the Digikey redirect. So this is most likely no issue in Part-DB itself.

Does your server (and PHP) with Part-DB have internet access or is there something intercepting traffic?
Have you given the API application the  access to the "Product Information" api on Digikey API config? Is this really a production application.

Maybe it helps, if you create a new fresh API application in Digikeys API portal.

ewa, 2023-09-13:
Okay, so I have a fresh (production) application set up on DigiKey with corresponding credentials, and I have completely-stock version of Part-DB `master` checked out, built and started with [this docker-compose.yaml](https://docs.part-db.de/installation/installation_docker.html) file, modified only by adding the secret and client id variables.  It's running on an Azure VM in the cloud behind an http/https proxy.  

Now, instead of the prior error, I get a new error, directly from DigiKey:
`{"ErrorCode" : "invalid_request", "Error" :"Invalid redirection uri http://ewa-gw.westus3.cloudapp.azure.com/en/oauth/client/ip_digikey_oauth/check"}`

Note that the uri begins **`http`**, not **`https`**.  I'm not sure where or why Part-DB is setting that to `http`; the URL I'm accessing to make the request is `https://ewa-gw.westus3.cloudapp.azure.com/en/tools/info_providers/providers` (and the DigiKey endpoint URL is `
https://ewa-gw.westus3.cloudapp.azure.com/en/oauth/client/ip_digikey_oauth/check`).

It's probably relevant that (like pretty much all proxy setups), my setup is using https from the end user to the proxy (`ewa-gw.westus3.clouadapp.azure.com`) and then http from the proxy to the virtual machine and docker container.  I'm not sure how to tell Part-DB that it's really running https:  I tried passing in      "` - DEFAULT_URI=https://ewa-gw.westus3.cloudapp.azure.com/`" from the compose file, but that doesn't change anything.

Any suggestions?

Also, when you tested it, did you try the "Connect OAuth" process with fresh credentials?  Is there any chance that it's working for you because of some cached credentials/state?

ewa, 2023-09-13:
Setting   `TRUSTED_PROXIES=10.0.1.0/24` where that's the (private IP) subnet for my proxy server(s) seems to make it work.

So, I would update the "bug" report from "it doesn't work" to "it can work, but needs much better diagnostic messages about what's wrong when it doesn't work."

closed 2023-11-19 by jbtronics