Part-DB/Part-DB-server #329: Mouser and Octopart API

pdo59, 2023-07-21:
Is it possibile to add the Mouser and Octopart API? The Octopart API is paid, but there is also a free version which, although limited, is fine for those who do not make thousands of queries.

jbtronics, 2023-07-31:
Octopart provider is now available

mansellrace, 2023-08-01:
> Octopart provider is now available

Hi! i'm on 1.6.0 but i don't understand how to import from octopart api

jbtronics, 2023-08-01:
You have to setup an API key on Nexar and configure it at Part-DB like described here: https://docs.part-db.de/usage/information_provider_system.html

Afterwards you can use the octopart provider like any other provider from the "tools -> create part from info provider" page. (To be able to see and access that you have to give your users the permission for that on the user admin page).

RobThree, 2023-08-22:
Since this issue isn't closed yet, I thought I'd ask / provide feedback here.

I have all providers (TME, Digikey, Farnell Element14 and Octopart) configured and all work fine except Octopart. I'm running Part-DB Version 1.7.0-dev (master/098fcb2).

The prod.log contains:

```
{"message":"Matched route \"info_providers_search\".","context":{"route":"info_providers_search","route_parameters":{"_route":"info_providers_search","_controller":"App\\Controller\\InfoProviderController::search","_locale":"en"},"request_uri":"https://partdb.mydomain.nl/en/tools/info_providers/search","method":"POST"},"level":200,"level_name":"INFO","channel":"request","datetime":"2023-08-22T03:14:25.472416+02:00","extra":{"token":null,"url":"/en/tools/info_providers/search","ip":"x.x.x.x","http_method":"POST","server":"partdb.home.mydomain.nl","referrer":"https://partdb.mydomain.nl/en/tools/info_providers/search"}}
{"message":"Read existing security token from the session.","context":{"key":"_security_main","token_class":"Symfony\\Component\\Security\\Core\\Authentication\\Token\\RememberMeToken"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-08-22T03:14:25.476935+02:00","extra":{"token":null,"url":"/en/tools/info_providers/search","ip":"x.x.x.x","http_method":"POST","server":"partdb.home.mydomain.nl","referrer":"https://partdb.mydomain.nl/en/tools/info_providers/search"}}
{"message":"User Deprecated: Using XML mapping driver with XSD validation disabled is deprecated and will not be supported in Doctrine ORM 3.0. (XmlDriver.php:60 called by App_KernelProdContainer.php:296, https://github.com/doctrine/orm/pull/6728, package doctrine/orm)","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Using XML mapping driver with XSD validation disabled is deprecated and will not be supported in Doctrine ORM 3.0. (XmlDriver.php:60 called by App_KernelProdContainer.php:296, https://github.com/doctrine/orm/pull/6728, package doctrine/orm)","code":0,"file":"/var/www/partdb/vendor/doctrine/deprecations/lib/Doctrine/Deprecations/Deprecation.php:209"}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-08-22T03:14:25.477337+02:00","extra":{"token":null,"url":"/en/tools/info_providers/search","ip":"x.x.x.x","http_method":"POST","server":"partdb.home.mydomain.nl","referrer":"https://partdb.mydomain.nl/en/tools/info_providers/search"}}
{"message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\EventLoggerSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] attribute.","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\EventLoggerSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] attribute.","code":0,"file":"/var/www/partdb/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:192"}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-08-22T03:14:25.481191+02:00","extra":{"token":null,"url":"/en/tools/info_providers/search","ip":"x.x.x.x","http_method":"POST","server":"partdb.home.mydomain.nl","referrer":"https://partdb.mydomain.nl/en/tools/info_providers/search"}}
{"message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\LogDBMigrationSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] attribute.","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\LogDBMigrationSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] attribute.","code":0,"file":"/var/www/partdb/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:192"}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-08-22T03:14:25.481517+02:00","extra":{"token":null,"url":"/en/tools/info_providers/search","ip":"x.x.x.x","http_method":"POST","server":"partdb.home.mydomain.nl","referrer":"https://partdb.mydomain.nl/en/tools/info_providers/search"}}
{"message":"User Deprecated: Subscribing to postConnect events is deprecated. Implement a middleware instead. (Connection.php:385 called by Connection.php:450, https://github.com/doctrine/dbal/issues/5784, package doctrine/dbal)","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Subscribing to postConnect events is deprecated. Implement a middleware instead. (Connection.php:385 called by Connection.php:450, https://github.com/doctrine/dbal/issues/5784, package doctrine/dbal)","code":0,"file":"/var/www/partdb/vendor/doctrine/deprecations/lib/Doctrine/Deprecations/Deprecation.php:209"}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-08-22T03:14:25.481693+02:00","extra":{"token":null,"url":"/en/tools/info_providers/search","ip":"x.x.x.x","http_method":"POST","server":"partdb.home.mydomain.nl","referrer":"https://partdb.mydomain.nl/en/tools/info_providers/search"}}
{"message":"User was reloaded from a user provider.","context":{"provider":"Symfony\\Bridge\\Doctrine\\Security\\User\\EntityUserProvider","username":"admin"},"level":100,"level_name":"DEBUG","channel":"security [... cut ...] h","ip":"x.x.x.x","http_method":"POST","server":"partdb.home.mydomain.nl","referrer":"https://partdb.mydomain.nl/en/tools/info_providers/search"}}
```

I'm quite sure I entered the authorization correctly:

[image]

[image]

I also tried `DE` and `US` countries, no difference. The provided IDE seems to work fine, returning results:

[image]

Octopart/Nexus seems to return a 500 Internal Server error - indicating something may be wrong with the request?

I hope I'm not wasting your time *again* 😬

jbtronics, 2023-08-22:
On my instance it works fine.
Can you look into the database and check if there is any entry in the `oauth_tokens` table?

RobThree, 2023-08-22:
>  Can you look into the database and check if there is any entry in the oauth_tokens table?

Ha! There was! I deleted the Octopart record and tried again and it worked. I guess my first attempt had worked in the sense that it had gotten an OAuth token but then failed because of I wasn't using the "Evaluation app" but rather a (limited to 0 requests) application. I then changed the ClientId/Secret to the correct application but, ofcourse, the OAuth token was still from the previous attempt. Had I known there was an `oauth_tokens` table and the providers stored tokens there I would've removed it myself. Thanks! Again! 👍

shadowlab, 2023-10-06:
Good evening, 

Was there any possible update with the mouser API? They offer free accounts with a 1000 result per day limit via signup at the following: 

https://www.mouser.com/api-search/ 

Thanks for all your work so far!

pdo59, 2023-10-06:
Hi shadowlab,
This week I'll post the code for the Mouser API to Jan. I've tested it and it seems to work. The only limitation is that the Mouser API does not offer any parameters for parts (eg. resistance, capacitance, ecc.).

jbtronics, 2023-10-07:
I added @pdo59 MouserProvider with some slight modifications to the Part-DB master branch.
Thank you very much @pdo59

closed 2023-10-07 by jbtronics