Part-DB/Part-DB-server #331: Automatic parts update

pdo59, 2023-07-23:
I have migrated many parts from my old inventory software and for all of these I have no parameters, images, datasheets etc.

It would be very useful to be able to update this information using the APIs of the distributors. You could add an action in the "Tools" tab of each part to operate individually and manually.
Then a special flag (like the one already present "Needs review") called "Update from providers" would allow the selection of the parts to be updated with a special tool. This tool may list parts marked "Update from providers" and allow them to be updated automatically.

pdo59, 2023-09-25:
I propose this topic again because I consider it very useful: in my DB many parts, imported from previous software, are without important data such as parameters, datasheets, images etc. It would be really useful to be able to complete this information through data providers.

jbtronics, 2023-11-25:
With Part-DB 1.9.0 you can now update existing parts using the info providers. 
It's not automatic, as you still need to review the changes and decide what to keep, but making the complete process automatic would be very difficult.

pdo59, 2023-11-26:
This is exactly what I thought! This makes upgrading a component board much easier. 
Thank's jbtronics!