Part-DB/Part-DB-server #332: 500 Internal Server Error

blue3121, 2023-07-24:
Updating to version (without errors)

1.6.0-dev (master/be2ed52) 

browsing the categories I get the error:

**500 Internal Server Error**

I have tried to follow all the instructions provided to no avail !!!

______________________________________________________

- PHP Version: 8.1.2-1ubuntu2.13
- Database Server: SqLite

 - OS: Kubuntu 22.04
 - Browser: Firefox
 - Version: 115.0.2 (64 bit)

blue3121, 2023-07-24:
{"message":"Matched route \"redirector\".","context":{"route":"redirector","route_parameters":{"_route":"redirector","_controller":"App\\Controller\\RedirectController::addLocalePart","url":""},"request_uri":"http://127.0.0.1/","method":"GET"},"level":200,"level_name":"INFO","channel":"request","datetime":"2023-04-01T18:25:44.528414+02:00","extra":{"token":null,"url":"/","ip":"127.0.0.1","http_method":"GET","server":"127.0.0.1","referrer":null}}
{"message":"Checking for authenticator support.","context":{"firewall_name":"main","authenticators":4},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-04-01T18:25:44.565371+02:00","extra":{"token":null,"url":"/","ip":"127.0.0.1","http_method":"GET","server":"127.0.0.1","referrer":null}}
{"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Hslavich\\OneloginSamlBundle\\Security\\Http\\Authenticator\\SamlAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-04-01T18:25:44.565388+02:00","extra":{"token":null,"url":"/","ip":"127.0.0.1","http_method":"GET","server":"127.0.0.1","referrer":null}}
{"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Hslavich\\OneloginSamlBundle\\Security\\Http\\Authenticator\\SamlAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-04-01T18:25:44.565395+02:00","extra":{"token":null,"url":"/","ip":"127.0.0.1","http_method":"GET","server":"127.0.0.1","referrer":null}}
{"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\FormLoginAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-04-01T18:25:44.565399+02:00","extra":{"token":null,"url":"/","ip":"127.0.0.1","http_method":"GET","server":"127.0.0.1","referrer":null}}
{"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\FormLoginAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-04-01T18:25:44.565404+02:00","extra":{"token":null,"url":"/","ip":"127.0.0.1","http_method":"GET","server":"127.0.0.1","referrer":null}}
{"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Scheb\\TwoFactorBundle\\Security\\Http\\Authenticator\\TwoFactorAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-04-01T18:25:44.565408+02:00","extra":{"token":null,"url":"/","ip":"127.0.0.1","http_method":"GET","server":"127.0.0.1","referrer":null}}
{"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Scheb\\TwoFactorBundle\\Security\\Http\\Authenticator\\TwoFactorAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-04-01T18:25:44.565414+02:00","extra":{"token":null,"url":"/","ip":"127.0.0.1","http_method":"GET","server":"127.0.0.1","referrer":null}}
{"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\RememberMeAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-04-01T18:25:44.565417+02:00","extra":{"token":null,"url":"/","ip":"127.0.0.1","http_method":"GET","server":"127.0.0.1","referrer":null}}
{"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\RememberMeAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-04-01T18:25:44.565545+02:00","extra":{"token":null,"url":"/","ip":"127.0.0.1","http_method":"GET","server":"127.0.0.1","referrer":null}}
{"message":"User Deprecated: Since symfony/framework-bundle 5.1: The \"session.flash_bag\" service is deprecated, use \"$session->getFlashBag()\" instead.","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Since symfony/framework-bundle 5.1: The \"session.flash_bag\" service is deprecated, use \"$session->getFlashBag()\" instead.","code":0,"file":"/var/www/partdb/var/cache/prod/ContainerOEdhxIV/getSession_FlashBagService.php:22"}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-04-01T18:25:44.572334+02:00","extra":{"token":null,"url":"/","ip":"127.0.0.1","http_method":"GET","server":"127.0.0.1","referrer":null}}
{"message":"Deprecated: Return type of App\\Entity\\UserSystem\\PermissionData::jsonSerialize() should either be compatible with JsonSerializable::jsonSerialize(): mixed, or the #[\\ReturnTypeWillChange] attribute should be used to temporarily suppress the notice","context":{"exception":{"class":"ErrorException","message":"Deprecated: Return type of App\\Entity\\UserSystem\\PermissionData::jsonSerialize() should either be compatible with JsonSerializable::jsonSerialize(): mixed, or the #[\\ReturnTypeWillChange] attribute should be used to temporarily suppress the notice","code":0,"file":"/var/www/partdb/src/Entity/UserSystem/PermissionData.php:208"}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-04-01T18:25:44.664425+02:00","extra":{"token":null,"url":"/","ip":"127.0.0.1","http_method":"GET","server":"127.0.0.1","referrer":null}}
[... cut ...] ine/dbal/src/Driver/PDO/Exception.php:28","previous":{"class":"PDOException","message":"SQLSTATE[HY000]: General error: 1 too many arguments on function FIELD","code":0,"file":"/var/www/partdb/vendor/doctrine/dbal/src/Driver/PDO/Connection.php:58"}}}},"level":500,"level_name":"CRITICAL","channel":"request","datetime":"2023-07-24T14:02:38.397942+02:00","extra":{"token":null,"url":"/en/category/230/parts","ip":"127.0.0.1","http_method":"GET","server":"127.0.0.1","referrer":"http://127.0.0.1/en/"}}

jbtronics, 2023-07-24:
You should try to clear the cache with `sudo -u www-data php bin/console cache:clear`.

And do a chown on the part-db files `chown www-data:www-data -R *`

blue3121, 2023-07-24:
tried everything, it doesn't work

jbtronics, 2023-07-27:
This is fixed in the latest master version.

closed 2023-07-27 by jbtronics

blue3121, 2023-07-27:
Good job !!!
Thank you ! 
Now works !!!