Part-DB/Part-DB-server #333: Import assistant for partkeepr throws critical error and stops

nodurah, 2023-07-25:
Hello,

first of all thanks for addressing the import/export issue to make the transition from partkeepr to part-db less painful! This feature was the missing stone to migrate to your project.

Unfortunately the migration assistant exits in the "Importing parts" stage with an error while importing the .xml data:

`docker exec -it a8f3b82d5c14 php bin/console partdb:migrations:import-partkeepr /test.xml`

```
{"message":"Error thrown while running command \"partdb:migrations:import-partkeepr '/test.xml'\". Message: \"Could not find manufacturer with ID \"","context":{"exception":{"class":"RuntimeException","message":"Could not find manufacturer with ID ","code":0,"file":"/var/www/html/src/Services/ImportExportSystem/PartKeeprImporter/PKPartImporter.php:149","trace":["/var/www/html/src/Services/ImportExportSystem/PartKeeprImporter/PKPartImporter.php:123","/var/www/html/src/Command/Migrations/ImportPartKeeprCommand.php:142","/var/www/html/src/Command/Migrations/ImportPartKeeprCommand.php:95","/var/www/html/vendor/symfony/console/Command/Command.php:326","/var/www/html/vendor/symfony/console/Application.php:1081","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:91","/var/www/html/vendor/symfony/console/Application.php:320","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:80","/var/www/html/vendor/symfony/console/Application.php:174","/var/www/html/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:54","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/bin/console:11"]},"command":"partdb:migrations:import-partkeepr '/test.xml'","message":"Could not find manufacturer with ID "},"level":500,"level_name":"CRITICAL","channel":"console","datetime":"2023-07-25T19:25:43.223257+00:00","extra":{"token":null,"command":{"name":"partdb:migrations:import-partkeepr","arguments":{"command":"partdb:migrations:import-partkeepr","file":"/test.xml"}}}}
19:25:43 CRITICAL  [console] Error thrown while running command "partdb:migrations:import-partkeepr '/test.xml'". Message: "Could not find manufacturer with ID " ["exception" => RuntimeException { …},"command" => "partdb:migrations:import-partkeepr '/test.xml'","message" => "Could not find manufacturer with ID "] ["token" => null,"command" => ["name" => "partdb:migrations:import-partkeepr","arguments" => ["command" => "partdb:migrations:import-partkeepr","file" => "/test.xml"]]]
{"message":"Command \"partdb:migrations:import-partkeepr '/test.xml'\" exited with code \"1\"","context":{"command":"partdb:migrations:import-partkeepr '/test.xml'","code":1},"level":100,"level_name":"DEBUG","channel":"console","datetime":"2023-07-25T19:25:43.230462+00:00","extra":{"token":null,"command":{"name":"partdb:migrations:import-partkeepr","arguments":{"command":"partdb:migrations:import-partkeepr","file":"/test.xml"}}}}

In PKPartImporter.php line 149:
                                       
  Could not find manufacturer with ID  
                                       

partdb:migrations:import-partkeepr [--no-projects] [--import-users] [--] <file>
```

In the partkeepr database I don't have an entry for this category for every part. As the import stops the imported data also seems to be corrupt and its not possible to log into part-db and view the data after a

`docker exec -it a7a2bb8e3ba1 php bin/console cache:clear`

Any help would be appreciated.

Kind regards and thanks in advance, 
Nodurah

jbtronics, 2023-07-25:
What Part-DB Version do you use?
What database type (MySQL or SQLite)?

If you are using SQLite it might help to remove the app.db file and then do the database migrations again (and don't create any new data using the webUI before you start the partkeepr import).

nodurah, 2023-07-25:
Hello jbtronics,

thanks for the quick reply. I forgot to mention that I'm using Version: 1.5.2 with squlite

A quick shutdown of the container, removing app.db restarting the container and starting the database migrations didn't have an effect.

Cheers, Nodurah

jbtronics, 2023-07-25:
For debugging it would be helpful, if you could send me the database dump (you can delete/censor user data or other sensitive informations). I guess that there is some edge case in your database, which is not handled correctly yet.

closed 2023-07-27 by jbtronics