Part-DB/Part-DB-server #334: PartKeepr to Part-DB migration in docker

myval, 2023-07-28:
Hi all,
I would like to use Part-DB but I am too dumb to manage the migration process from PartKeepr.
I run PartKeepr in Docker (since that was the easiest to get running) I think I managed to install and setup Part-DB in docker too but I struggle to import my data.
Managed to export PartKeepr database.
Can not figure out where all the attachments and pictures are stored, I am using the mhubig/partkeepr version
I got stucked at item 3 in the Part-DB how to migrate (https://docs.part-db.de/partkeepr_migration.html)

can anyone write some simple migration in docker for dummies that would be highly appreciated.

thanks
Josef

jbtronics, 2023-08-08:
basically you have to run the given command in your docker container. Should be possible with ´docker exec --user=www-data partdb php bin/console partdb:migrations:import-partkeepr path/to/pk.xml`, where `partdb` is the name of your docker container.

myval, 2023-08-19:
Thank you for your response, but I still struggle. When I run the command you suggested it spits out this:

  ! [NOTE] This command is still in development. If you encounter any problems,
 !        please report them to the issue tracker on GitHub.

 [WARNING] This command will delete all existing data in the database (except
           users). Make sure that you have no important data in the database
           before you continue!

 Please type "DELETE ALL DATA" to continue. []:
 > {"message":"Error thrown while running command \"partdb:migrations:import-partkeepr 'pk.xml'\". Message: \"You did not type \"DELETE ALL DATA\"!\"","context":{"exception":{"class":"RuntimeException","message":"You did not type \"DELETE ALL DATA\"!","code":0,"file":"/var/www/html/src/Command/Migrations/ImportPartKeeprCommand.php:73","trace":["/var/www/html/vendor/symfony/console/Helper/QuestionHelper.php:173","/var/www/html/vendor/symfony/console/Helper/QuestionHelper.php:77","/var/www/html/vendor/symfony/console/Style/SymfonyStyle.php:346","/var/www/html/vendor/symfony/console/Style/SymfonyStyle.php:257","/var/www/html/src/Command/Migrations/ImportPartKeeprCommand.php:76","/var/www/html/vendor/symfony/console/Command/Command.php:326","/var/www/html/vendor/symfony/console/Application.php:1081","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:91","/var/www/html/vendor/symfony/console/Application.php:320","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:80","/var/www/html/vendor/symfony/console/Application.php:174","/var/www/html/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:54","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/bin/console:11"]},"command":"partdb:migrations:import-partkeepr 'pk.xml'","message":"You did not type \"DELETE ALL DATA\"!"},"level":500,"level_name":"CRITICAL","channel":"console","datetime":"2023-08-19T11:07:35.651836+00:00","extra":{"token":null,"command":{"name":"partdb:migrations:import-partkeepr","arguments":{"command":"partdb:migrations:import-partkeepr","file":"pk.xml"}}}}
11:07:35 CRITICAL  [console] Error thrown while running command "partdb:migrations:import-partkeepr 'pk.xml'". Message: "You did not type "DELETE ALL DATA"!" ["exception" => RuntimeException { …},"command" => "partdb:migrations:import-partkeepr 'pk.xml'","message" => "You did not type "DELETE ALL DATA"!"] ["token" => null,"command" => ["name" => "partdb:migrations:import-partkeepr","arguments" => ["command" => "partdb:migrations:import-partkeepr","file" => "pk.xml"]]]
{"message":"Command \"partdb:migrations:import-partkeepr 'pk.xml'\" exited with code \"1\"","context":{"command":"partdb:migrations:import-partkeepr 'pk.xml'","code":1},"level":100,"level_name":"DEBUG","channel":"console","datetime":"2023-08-19T11:07:35.659309+00:00","extra":{"token":null,"command":{"name":"partdb:migrations:import-partkeepr","arguments":{"command":"partdb:migrations:import-partkeepr","file":"pk.xml"}}}}

In ImportPartKeeprCommand.php line 73:

  You did not type "DELETE ALL DATA"!

partdb:migrations:import-partkeepr [--no-projects] [--import-users] [--] <file>

it has something about typing DELETE ALL DATA but where and when do I type it?

myval, 2023-08-19:
I am thinking I am not understanding the Path/To/ pk.xml
the pk.xml is in this folder E:\PART-DB and I run the command from this folder

If I use this command: E:\PART-DB>docker exec -it partdb php bin/console partdb:migrations:import-partkeepr pk.xml

I get this message
 ! [NOTE] This command is still in development. If you encounter any problems, please report them to the issue tracker
 !        on GitHub.

 [WARNING] This command will delete all existing data in the database (except users). Make sure that you have no
           important data in the database before you continue!

 Please type "DELETE ALL DATA" to continue. []:
 > DELETE ALL DATA

16:37:17 WARNING   [php] Warning: file_get_contents(pk.xml): Failed to open stream: No such file or directory ["exception" => ErrorException { …}] ["token" => null,"command" => ["name" => "partdb:migrations:import-partkeepr","arguments" => ["command" => "partdb:migrations:import-partkeepr","file" => "pk.xml"]]]
{"message":"User Deprecated: Using XML mapping driver with XSD validation disabled is deprecated and will not be supported in Doctrine ORM 3.0. (XmlDriver.php:65 called by App_KernelDockerContainer.php:290, https://github.com/doctrine/orm/pull/6728, package doctrine/orm)","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Using XML mapping driver with XSD validation disabled is deprecated and will not be supported in Doctrine ORM 3.0. (XmlDriver.php:65 called by App_KernelDockerContainer.php:290, https://github.com/doctrine/orm/pull/6728, package doctrine/orm)","code":0,"file":"/var/www/html/vendor/doctrine/deprecations/lib/Doctrine/Deprecations/Deprecation.php:209","trace":["/var/www/html/vendor/doctrine/deprecations/lib/Doctrine/Deprecations/Deprecation.php:107","/var/www/html/vendor/doctrine/orm/lib/Doctrine/ORM/Mapping/Driver/XmlDriver.php:65","/var/www/html/var/cache/docker/ContainerExkoa7x/App_KernelDockerContainer.php:290","/var/www/html/var/cache/docker/ContainerExkoa7x/App_KernelDockerContainer.php:283","/var/www/html/vendor/symfony/var-exporter/Internal/LazyObjectState.php:97","/var/www/html/vendor/symfony/var-exporter/LazyGhostTrait.php:176","/var/www/html/vendor/doctrine/orm/lib/Doctrine/ORM/EntityManager.php:207","/var/www/html/src/Doctrine/Purger/ResetAutoIncrementORMPurger.php:116","/var/www/html/src/Services/ImportExportSystem/PartKeeprImporter/PKImportHelper.php:46","/var/www/html/src/Command/Migrations/ImportPartKeeprCommand.php:82","/var/www/html/vendor/symfony/console/Command/Command.php:326","/var/www/html/vendor/symfony/console/Application.php:1081","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:91","/var/www/html/vendor/symfony/console/Application.php:320","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:80","/var/www/html/vendor/symfony/console/Application.php:174","/var/www/html/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:54","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/bin/console:11"]}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-08-19T16:37:15.214868+00:00","extra":{"token":null,"command":{"name":"partdb:migrations:import-partkeepr","arguments":{"command":"partdb:migrations:import-partkeepr","file":"pk.xml"}}}}
{"message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Using Doctrine subscribers as services is deprecated, declare listeners instead","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Using Doctrine subscribers as services is deprecated, declare listeners instead","code":0,"file":"/var/www/html/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:192","trace":["/var/www/html/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:101","/var/www/html/vendor/doctrine/dbal/src/Connection.php:384","/var/www/html/vendor/doctrine/dbal/src/Connection.php:450","/var/www/html/vendor/doctrine/dbal/src/Connection.php:408","/var/www/html/vendor/doctrine/dbal/src/Connection.php:316","/var/www/html/src/Doctrine/Purger/ResetAutoIncrementORMPurger.php:127","/var/www/html/src/Services/ImportExportSystem/PartKeeprImporter/PKImportHelper.php:46","/var/www/html/src/Command/Migrations/ImportPartKeeprCommand.php:82","/var/www/html/vendor/symfony/console/Command/Command.php:326","/var/www/html/vendor/symfony/console/Application.php:1081","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:91","/var/www/html/vendor/symfony/console/Application.php:320","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:80","/var/www/html/vendor/symfony/console/Application.php:174","/var/www/html/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:54","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/bin/console:11"]}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-08-19T16:37:15.277701+00:00","extra":{"token":null,"command":{"name":"partdb:migrations:import-partkeepr","arguments":{"command":"partdb:migrations:import-partkeepr","file":"pk.xml"}}}}
{"message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Using Doctrine subscribers as services is deprecated, declare listeners instead","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Using Doctrine subscribers as services is deprecated, declare listeners instead","code":0,"file":"/var/www/html/vendor/symfony/doctrine-bridg [... cut ...] db:migrations:import-partkeepr","arguments":{"command":"partdb:migrations:import-partkeepr","file":"pk.xml"}}}}

In MySQLDumpXMLConverter.php line 48:

  App\Services\ImportExportSystem\PartKeeprImporter\MySQLDumpXMLConverter::convertMySQLDumpXMLDataToArrayStructure(): Argument #1 ($xml_string) must be of type string, bool given, called in /var/www/html/src/Command/Migrations/Import
  PartKeeprCommand.php on line 86

partdb:migrations:import-partkeepr [--no-projects] [--import-users] [--] <file>

jbtronics, 2023-08-19:
Put the `pk.xml` into the uploads folder mapped into the docker container (in the default docker-compose.yaml this should be the `upload` folder in the directory where the docker-compose.yaml lives). Then you can execute `docker exec -it partdb php bin/console partdb:migrations:import-partkeepr /var/www/html/uploads/pk.xml`

myval, 2023-08-19:
yeah thanks it is getting better but still getting some errors and there are no components imported at all:

```
[INFO] Importing distributors...

 [OK] Imported 2 distributors.

 [INFO] Importing part measurement units...

 [OK] Imported 1 measurement units.

 [INFO] Importing manufacturers...

 [OK] Imported 285 manufacturers.

 [INFO] Importing categories...

 [OK] Imported 156 categories.

 [INFO] Importing Footprints...

 [OK] Imported 189 footprints.

 [INFO] Importing storage locations...

17:43:51 WARNING   [php] Warning: Undefined array key "footprint_id" ["exception" => ErrorException { …}] ["token" => null,"command" => ["name" => "partdb:migrations:import-partkeepr","arguments" => ["command" => "partdb:migrations:import-partkeepr","file" => "/var/www/html/uploads/pk.xml"]]]
{"message":"User Deprecated: Using XML mapping driver with XSD validation disabled is deprecated and will not be supported in Doctrine ORM 3.0. (XmlDriver.php:65 called by App_KernelDockerContainer.php:290, https://github.com/doctrine/orm/pull/6728, package doctrine/orm)","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Using XML mapping driver with XSD validation disabled is deprecated and will not be supported in Doctrine ORM 3.0. (XmlDriver.php:65 called by App_KernelDockerContainer.php:290, https://github.com/doctrine/orm/pull/6728, package doctrine/orm)","code":0,"file":"/var/www/html/vendor/doctrine/deprecations/lib/Doctrine/Deprecations/Deprecation.php:209","trace":["/var/www/html/vendor/doctrine/deprecations/lib/Doctrine/Deprecations/Deprecation.php:107","/var/www/html/vendor/doctrine/orm/lib/Doctrine/ORM/Mapping/Driver/XmlDriver.php:65","/var/www/html/var/cache/docker/ContainerExkoa7x/App_KernelDockerContainer.php:290","/var/www/html/var/cache/docker/ContainerExkoa7x/App_KernelDockerContainer.php:283","/var/www/html/vendor/symfony/var-exporter/Internal/LazyObjectState.php:97","/var/www/html/vendor/symfony/var-exporter/LazyGhostTrait.php:176","/var/www/html/vendor/doctrine/orm/lib/Doctrine/ORM/EntityManager.php:207","/var/www/html/src/Doctrine/Purger/ResetAutoIncrementORMPurger.php:116","/var/www/html/src/Services/ImportExportSystem/PartKeeprImporter/PKImportHelper.php:46","/var/www/html/src/Command/Migrations/ImportPartKeeprCommand.php:82","/var/www/html/vendor/symfony/console/Command/Command.php:326","/var/www/html/vendor/symfony/console/Application.php:1081","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:91","/var/www/html/vendor/symfony/console/Application.php:320","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:80","/var/www/html/vendor/symfony/console/Application.php:174","/var/www/html/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:54","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/bin/console:11"]}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-08-19T17:43:44.165939+00:00","extra":{"token":null,"command":{"name":"partdb:migrations:import-partkeepr","arguments":{"command":"partdb:migrations:import-partkeepr","file":"/var/www/html/uploads/pk.xml"}}}}
{"message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Using Doctrine subscribers as services is deprecated, declare listeners instead","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Using Doctrine subscribers as services is deprecated, declare listeners instead","code":0,"file":"/var/www/html/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:192","trace":["/var/www/html/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:101","/var/www/html/vendor/doctrine/dbal/src/Connection.php:384","/var/www/html/vendor/doctrine/dbal/src/Connection.php:450","/var/www/html/vendor/doctrine/dbal/src/Connection.php:408","/var/www/html/vendor/doctrine/dbal/src/Connection.php:316","/var/www/html/src/Doctrine/Purger/ResetAutoIncrementORMPurger.php:127","/var/www/html/src/Services/ImportExportSystem/PartKeeprImporter/PKImportHelper.php:46","/var/www/html/src/Command/Migrations/ImportPartKeeprCommand.php:82","/var/www/html/vendor/symfony/console/Command/Command.php:326","/var/www/html/vendor/symfony/console/Application.php:1081","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:91","/var/www/html/vendor/symfony/console/Application.php:320","/var/www/html/vendor/symfony/framework-bundle/Console/Application.php:80","/var/www/html/vendor/symfony/console/Application.php:174","/var/www/html/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:54","/var/www/html/vendor/autoload_runtime.php:29","/var/www/html/bin/console:11"]}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-08-19T17:43:44.228252+00:00","extra":{"token":null,"command":{"name":"partdb:migrations:import-partkeepr","arguments":{"command":"partdb:migrations:import-partkeepr","file":"/var/www/html/uploads/pk.xml"}}}}
{"message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Using Doctrine subscribers as services is deprecated, declare listeners instead","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Using Doctrine subscribers as services is deprecated, declare listeners instead","code":0,"file":"/var/www/html/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:192","trace":["/var/www/html/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:101","/ [... cut ...]
{"message":"Command \"partdb:migrations:import-partkeepr '/var/www/html/uploads/pk.xml'\" exited with code \"1\"","context":{"command":"partdb:migrations:import-partkeepr '/var/www/html/uploads/pk.xml'","code":1},"level":100,"level_name":"DEBUG","channel":"console","datetime":"2023-08-19T17:43:51.820965+00:00","extra":{"token":null,"command":{"name":"partdb:migrations:import-partkeepr","arguments":{"command":"partdb:migrations:import-partkeepr","file":"/var/www/html/uploads/pk.xml"}}}}

In PKImportHelperTrait.php line 148:

  Could not find target entity with ID 0
```

jbtronics, 2023-08-19:
To debug that I will most likely need your pk.xml file.

You can censor sensitive information in it.

myval, 2023-08-19:
I don't think there is anything sensitive in it.
here it is:

[pk.zip](https://github.com/Part-DB/Part-DB-server/files/12386441/pk.zip)

jbtronics, 2023-08-19:
The import issue was related to the attachments of the storage locations. This is nox fixed.

The fix will be included in the next release (most likely 1.7.0). For now you can use the docker image tagged with the `edge` tag. (so replace the `image: jbtronics/part-db1:latest` with `image: jbtronics/part-db1:edge` and run a `docker-compose pull` and `docker-compose up` afterwards).

myval, 2023-08-20:
Yes this works like a charm, thanks.

closed 2023-08-21 by jbtronics