Part-DB/Part-DB-server #336: Store locations not showing up in edit menu

Lopfi, 2023-07-31:
**Describe the bug**
I imported data from an old inventory management system into Part-DB. Our System consists of about 150 boxes, each with 10 compartments. I wanted the exact location for each part e.g. box/comp (144/5) to show in the search without having to click on the part itself.
The problem is that this created about 1500 storage locations which is apparently too many to show in the list view of the edit tab.

**Server Side**
- Part-DB Version: Part-DB 1.0.1
- Database Server: SQLite

**Desktop (please complete the following information):**
 - OS: Arch
 - Browser: Firefox
 - Version: latest

**Additional context**
I understand that this might be a very niche issue and won't be fixed anytime soon. But I just wanted to report this as a known issue.

jbtronics, 2023-07-31:
On which page does this happen exactly and which element? The sidebar on the storage locations edit page, or the storage selector on the part editor? How does the issue looks like (a screenshot would be nice).
Can you see any issue in the logs of Part-DB or the browser developer console?

Can you try to delete the cache by running `php bin/console cache:clear` and see if that change anything?

If you are really running Part-DB 1.0 then you should try to update it,

Lopfi, 2023-08-02:
[image]
 
By clearing the cache the problem was resolved. And I didn't even notice that I was on such an old version I'll definitely update asap.

Thank you very much for the quick response