Part-DB/Part-DB-server #343: Error 500 when logging in

mase76, 2023-08-08:
Hi,
I have updated from 1.5.2 to 1.6.1 today. After a login I get the internal server error 500. Cache clearing did not help.
Here's are the last lines of the log:

{"message":"Authenticator does not support the request.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\FormLoginAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-08-08T16:32:21.325387+00:00","extra":{"token":null,"url":"/de/","ip":"10.42.42.251","http_method":"GET","server":"","referrer":null}}
{"message":"Checking support on authenticator.","context":{"firewall_name":"main","authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\RememberMeAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-08-08T16:32:21.325413+00:00","extra":{"token":null,"url":"/de/","ip":"10.42.42.251","http_method":"GET","server":"","referrer":null}}
{"message":"Remember-me cookie detected.","context":{},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-08-08T16:32:21.325460+00:00","extra":{"token":null,"url":"/de/","ip":"10.42.42.251","http_method":"GET","server":"","referrer":null}}
{"message":"Read existing security token from the session.","context":{"key":"_security_main","token_class":"Symfony\\Component\\Security\\Core\\Authentication\\Token\\RememberMeToken"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-08-08T16:32:21.326682+00:00","extra":{"token":null,"url":"/de/","ip":"10.42.42.251","http_method":"GET","server":"","referrer":null}}
{"message":"User Deprecated: Using XML mapping driver with XSD validation disabled is deprecated and will not be supported in Doctrine ORM 3.0. (XmlDriver.php:60 called by App_KernelProdContainer.php:296, https://github.com/doctrine/orm/pull/6728, package doctrine/orm)","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Using XML mapping driver with XSD validation disabled is deprecated and will not be supported in Doctrine ORM 3.0. (XmlDriver.php:60 called by App_KernelProdContainer.php:296, https://github.com/doctrine/orm/pull/6728, package doctrine/orm)","code":0,"file":"/usr/local/src/Part-DB-server/vendor/doctrine/deprecations/lib/Doctrine/Deprecations/Deprecation.php:209"}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-08-08T16:32:21.327245+00:00","extra":{"token":null,"url":"/de/","ip":"10.42.42.251","http_method":"GET","server":"","referrer":null}}
{"message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\EventLoggerSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] attribute.","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\EventLoggerSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] attribute.","code":0,"file":"/usr/local/src/Part-DB-server/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:192"}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-08-08T16:32:21.331681+00:00","extra":{"token":null,"url":"/de/","ip":"10.42.42.251","http_method":"GET","server":"","referrer":null}}
{"message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\LogDBMigrationSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] attribute.","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Since symfony/doctrine-bridge 6.3: Registering \"App\\EventSubscriber\\LogSystem\\LogDBMigrationSubscriber\" as a Doctrine subscriber is deprecated. Register it as a listener instead, using e.g. the #[AsDoctrineListener] attribute.","code":0,"file":"/usr/local/src/Part-DB-server/vendor/symfony/doctrine-bridge/ContainerAwareEventManager.php:192"}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-08-08T16:32:21.332104+00:00","extra":{"token":null,"url":"/de/","ip":"10.42.42.251","http_method":"GET","server":"","referrer":null}}
{"message":"User Deprecated: Subscribing to postConnect events is deprecated. Implement a middleware instead. (Connection.php:385 called by Connection.php:450, https://github.com/doctrine/dbal/issues/5784, package doctrine/dbal)","context":{"exception":{"class":"ErrorException","message":"User Deprecated: Subscribing to postConnect events is deprecated. Implement a middleware instead. (Connection.php:385 called by Connection.php:450, https://github.com/doctrine/dbal/issues/5784, package doctrine/dbal)","code":0,"file":"/usr/local/src/Part-DB-server/vendor/doctrine/deprecations/lib/Doctrine/Deprecations/Deprecation.php:209"}},"level":200,"level_name":"INFO","channel":"deprecation","datetime":"2023-08-08T16:32:21.332391+00:00","extra":{"token":null,"url":"/de/","ip":"10.42.42.251","http_method":"GET","server":"","referrer":null}}
{"message":"User was reloaded from a user provider.","context":{"provider":"Symfony\\Bridge\\Doctrine\\Security\\User\\EntityUserProvider","username":"mase"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-08-08T16:32:21.336004+00:00","extra":{"token":null,"url":"/de/","ip":"10.42.42.251","http_method":"GET","server":"","referrer":null}}
{"message":"Skipping the \"Symfony\\Component\\Security\\Http\\Authenticator\\RememberMeAuthenticator\" authenticator as it did not support the request.","context":{"authenticator":"Symfony\\Component\\Security\\Http\\Authenticator\\RememberMeAuthenticator"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-08-08T16:32:21.336163+00:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"mase"},"url":"/de/","ip":"10.42.42.251","http_method":"GET","server":"","referrer":null}}
{"message":"Uncaught PHP Exception Twig\\Error\\RuntimeError: \"An exception has been thrown during the rendering of a template (\"Entity of type 'App\\Entity\\Attachments\\UserAttachment' for IDs id(3) was not found\").\" at /usr/local/src/Part-DB-server/templates/_navbar.html.twig line 53","context":{"exception":{"class":"Twig\\Error\\RuntimeError","message":"An exception has been thrown during the rendering of a template (\"Entity of type 'App\\Entity\\Attachments\\UserAttachment' for IDs id(3) was not found\").","code":0,"file":"/usr/local/src/Part-DB-server/templates/_navbar.html.twig:53","previous":{"class":"Doctrine\\ORM\\EntityNotFoundException","message":"Entity of type 'App\\Entity\\Attachments\\UserAttachment' for IDs id(3) was not found","code":0,"file":"/usr/local/src/Part-DB-server/vendor/doctrine/orm/lib/Doctrine/ORM/EntityNotFoundException.php:33"}}},"level":500,"level_name":"CRITICAL","channel":"request","datetime":"2023-08-08T16:32:21.358175+00:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"mase"},"url":"/de/","ip":"10.42.42.251","http_method":"GET","server":"","referrer":null}}
{"message":"Stored the security token in the session.","context":{"key":"_security_main"},"level":100,"level_name":"DEBUG","channel":"security","datetime":"2023-08-08T16:32:21.364371+00:00","extra":{"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"mase"},"url":"/de/","ip":"10.42.42.251","http_method":"GET","server":"","referrer":null}}

jbtronics, 2023-08-08:
Have you upgraded the database schema after you upgraded Part-DB?

It seems like it can not find the user profile picture (or more exactly the database element containing the profile picture). This looks like there is an invalid reference to a non-existing attachment. However that should not be possible, as foreign key constraints should forbid that.
Have you  somehow disabled foreign keys on your database key or manually edited Part-DB database?

mase76, 2023-08-08:
I did the migration. I restored the old database several times and migrated again. No luck.

mase76, 2023-08-08:
I altered id_preview_attachment of my user to NULL and now I can log in again.

mase76, 2023-08-08:
I noticed, that a distributor showed my profile picture as logo. I deleted the logo and created a new one. I hope, that was the only mess.

closed 2023-08-20 by jbtronics