Part-DB/Part-DB-server #351: Resized (cached?) images don't appear to work

RobThree, 2023-08-21:
**Describe the bug**
None of the resized images (thumbnails) seem to load properly.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to any image upload - in this case we'll try a user avatar
2. Upload the image
3. Notice that the thumbnail doesn't display

**Expected behavior**
I expect thumbnails to work properly

**Screenshots**
[image]

[image]

[image]
Note in the above image that a footprint (non-thumbnail) loads correctly

**Server Side**
- Part-DB Version: 1.7.0-dev
- PHP Version: 8.2.8
- Database Server: MariaDB 10.11.3-MariaDB-1
- Server Operating System: Debian 6.1.38-4 (2023-08-08) x86_64 (Bookworm)
- Opcache enabled: True
- Maximum upload size (upload_max_filesize / post_max_size): 50M / 50M
- PHP extensions: Core date libxml openssl pcre zlib filter hash json random Reflection SPL session standard sodium cgi-fcgi mysqlnd PDO xml apcu ast bcmath bz2 calendar ctype curl dom mbstring FFI fileinfo ftp gd gettext iconv igbinary imagick imap intl exif msgpack mysqli pdo_mysql pdo_sqlite Phar posix pspell readline shmop SimpleXML snmp soap sockets sqlite3 sysvmsg sysvsem sysvshm tidy tokenizer xmlreader xmlwriter xsl zip memcached Zend OPcache xdebug

**Desktop (please complete the following information):**
 - OS: MacOS Ventura 13.4.1
 - Browser: Chrome 116.0.5845.96

**Additional context**
All images under https://partdb.mydomain.nl/media/cache/... don't seem to load properly. The cache directory exists and has proper permissions AFAIK:

[image]

Uploads do seem to work, when I look in `/var/www/partdb/public/media/user/2` there is an image: `avatar-64e37182c18d8.jpeg`.

The links to the images return a 404.

Also, in the console, are some errors about fonts:

[image]

The `var/log/prod*` files don't seem to have any relevant info (I will gladly post them when required). The nginx log, however, seems to have something relevant:

```cmd
2023/08/21 16:02:32 [error] 393839#393839: *111048 open() "/var/www/partdb/public/media/cache/resolve/thumbnail_xs/img/footprints/Passive/Capacitors/Ceramic/SMD/Ceramic-SMD_0805.png" failed (2: No such file or directory), client: 192.168.7.254, server: partdb.home.mydomain.nl, request: "GET /media/cache/resolve/thumbnail_xs/img/footprints/Passive/Capacitors/Ceramic/SMD/Ceramic-SMD_0805.png HTTP/2.0", host: "partdb.mydomain.nl", referrer: "https://partdb.mydomain.nl/en/part/20/clone"
2023/08/21 16:02:32 [error] 393839#393839: *111048 open() "/var/www/partdb/public/media/cache/resolve/thumbnail_xs/img/footprints/Passive/Capacitors/Ceramic/SMD/Ceramic-SMD_1206.png" failed (2: No such file or directory), client: 192.168.7.254, server: partdb.home.mydomain.nl, request: "GET /media/cache/resolve/thumbnail_xs/img/footprints/Passive/Capacitors/Ceramic/SMD/Ceramic-SMD_1206.png HTTP/2.0", host: "partdb.mydomain.nl", referrer: "https://partdb.mydomain.nl/en/part/20/clone"
2023/08/21 16:02:32 [error] 393839#393839: *111048 open() "/var/www/partdb/public/media/cache/resolve/thumbnail_xs/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0603.png" failed (2: No such file or directory), client: 192.168.7.254, server: partdb.home.mydomain.nl, request: "GET /media/cache/resolve/thumbnail_xs/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0603.png HTTP/2.0", host: "partdb.mydomain.nl", referrer: "https://partdb.mydomain.nl/en/part/20/clone"
2023/08/21 16:02:32 [error] 393839#393839: *111048 open() "/var/www/partdb/public/media/cache/resolve/thumbnail_xs/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0805.png" failed (2: No such file or directory), client: 192.168.7.254, server: partdb.home.mydomain.nl, request: "GET /media/cache/resolve/thumbnail_xs/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0805.png HTTP/2.0", host: "partdb.mydomain.nl", referrer: "https://partdb.mydomain.nl/en/part/20/clone"
2023/08/21 16:02:38 [error] 393839#393839: *111048 open() "/var/www/partdb/public/media/cache/resolve/thumbnail_sm/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0402.png" failed (2: No such file or directory), client: 192.168.7.254, server: partdb.home.mydomain.nl, request: "GET /media/cache/resolve/thumbnail_sm/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0402.png HTTP/2.0", host: "partdb.mydomain.nl", referrer: "https://partdb.mydomain.nl/en/category/50/parts"
2023/08/21 16:02:49 [error] 393839#393839: *111048 open() "/var/www/partdb/public/media/cache/resolve/thumbnail_sm/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0402.png" failed (2: No such file or directory), client: 192.168.7.254, server: partdb.home.mydomain.nl, request: "GET /media/cache/resolve/thumbnail_sm/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0402.png HTTP/2.0", host: "partdb.mydomain.nl", referrer: "https://partdb.mydomain.nl/en/category/50/parts"
2023/08/21 16:02:50 [error] 393839#393839: *111048 open() "/var/www/partdb/public/media/cache/resolve/thumbnail_sm/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0402.png" failed (2: No such file or directory), client: 192.168.7.254, server: partdb.home.mydomain.nl, request: "GET /media/cache/resolve/thumbnail_sm/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0402.png HTTP/2.0", host: "partdb.mydomain.nl", referrer: "https://partdb.mydomain.nl/en/category/50/parts"
2023/08/21 16:04:17 [error] 393839#393839: *111048 open() "/var/www/partdb/public/media/cache/resolve/thumbnail_sm/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0402.png" failed (2: No such file or directory), client: 192.168.7.254, server: partdb.home.mydomain.nl, request: "GET /media/cache/resolve/thumbnail_sm/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0402.png HTTP/2.0", host: "partdb.mydomain.nl", referrer: "https://partdb.mydomain.nl/en/category/50/parts"
2023/08/21 16:04:18 [error] 393839#393839: *111048 open() "/var/www/partdb/public/media/cache/resolve/thumbnail_sm/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0402.png" failed (2: No such file or directory), client: 192.168.7.254, server: partdb.home.mydomain.nl, request: "GET /media/cache/resolve/thumbnail_sm/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0402.png HTTP/2.0", host: "partdb.mydomain.nl", referrer: "https://partdb.mydomain.nl/en/part/20/info"
2023/08/21 16:04:18 [error] 393839#393839: *111048 open() "/var/www/partdb/public/media/cache/resolve/thumbnail_sm/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0402.png" failed (2: No such file or directory), client: 192.168.7.254, server: partdb.home.mydomain.nl, request: "GET /media/cache/resolve/thumbnail_sm/img/footprints/Passive/Resistors/SMD/RESISTOR-SMD_0402.png HTTP/2.0", host: "partdb.mydomain.nl", referrer: "https://partdb.mydomain.nl/en/part/20/info"
2023/08/21 16:15:30 [error] 393837#393837: *113113 open() "/var/www/partdb/public/media/cache/resolve/thumbnail_md/media/user/2/avatar-64e37182c18d8.jpeg" failed (2: No such file or directory), client: 192.168.7.254, server: partdb.home.mydomain.nl, request: "GET /media/cache/resolve/thumbnail_md/media/user/2/avatar-64e37182c18d8.jpeg HTTP/2.0", host: "partdb.mydomain.nl", referrer: "https://partdb.mydomain.nl/en/user/settings"
2023/08/21 16:15:31 [error] 393837#393837: *113113 open() "/var/www/partdb/public/media/cache/resolve/thumbnail_xs/media/user/2/avatar-64e37182c18d8.jpeg" failed (2: No such file or directory), client: 192.168.7.254, server: partdb.home.mydomain.nl, request: "GET /media/cache/resolve/thumbnail_xs/media/user/2/avatar-64e37182c18d8.jpeg HTTP/2.0", host: "partdb.mydomain.nl", referrer: "https://partdb.mydomain.nl/en/user/settings"
2023/08/21 16:15:31 [error] 393837#393837: *113113 open() "/var/www/partdb/public/media/cache/resolve/thumbnail_md/media/user/2/avatar-64e37182c18d8.jpeg" failed (2: No such file or directory), client: 192.168.7.254, server: partdb.home.mydomain.nl, request: "GET /media/cache/resolve/thumbnail_md/media/user/2/avatar-64e37182c18d8.jpeg HTTP/2.0", host: "partdb.mydomain.nl", referrer: "https://partdb.mydomain.nl/en/user/settings"
2023/08/21 16:16:50 [error] 393837#393837: *113113 open() "/var/www/partdb/public/media/cache/resolve/thumbnail_xs/media/user/2/avatar-64e37182c18d8.jpeg" failed (2: No such file or directory), client: 192.168.7.254, server: partdb.home.mydomain.nl, request: "GET /media/cache/resolve/thumbnail_xs/media/user/2/avatar-64e37182c18d8.jpeg HTTP/2.0", host: "partdb.mydomain.nl", referrer: "https://partdb.mydomain.nl/en/category/50/parts"
[... cut ...]
        return 404;
    }
}
```

I'm not sure how to debug this issue; especially since I'm new to the project (discovered it yesterday, great work by the way!).

My guess is that the image-resizer and/or caching doesn't get hit properly; I don't expect it to be a permissions issue (oh, I've cleared the cache, per the instructions, a few good times while trying several things too).

If there's any more information you need, please let me know. I'll also be happy to provide a login to my instance if that helps.

jbtronics, 2023-08-21:
Do the cached thumbnails exist in the `public/media/cache` folder?
I would suspect that no thumbnails are generated (as nginx seems to not be able to find them). It is important that, if the server does not find the asset directly, that the request is instead mapped to the index.php, so that the thumbnail generator can create the thumbnail file. On subsequent access tries, the file is there and nginx can serve that file directly (without invoking PHP).
I guess your configuration somehow prevents that.

jbtronics, 2023-08-21:
Maybe [this](https://github.com/liip/LiipImagineBundle/issues/977) or [this](https://github.com/liip/LiipImagineBundle/issues/784#issuecomment-320150743) issue of the thumbnail generation library might help you.

RobThree, 2023-08-21:
> Do the cached thumbnails exist in the public/media/cache folder?

Nope, that's exactly why I figured the PHP doesn't probably even get hit.

> I guess your configuration somehow prevents that.

I guess so, but because everything else seems to work I wasn't sure... URL-rewriting seems to work; I tried setting the `NO_URL_REWRITE_AVAILABLE` to `1` to see if that changed anything, but then it stopped working entirely.

> Maybe [this](https://github.com/liip/LiipImagineBundle/issues/977) or [this](https://github.com/liip/LiipImagineBundle/issues/784#issuecomment-320150743) issue of the thumbnail generation library might help you.

I'll go check it out.

Is there any way we can see if the PHP gets hit even?

RobThree, 2023-08-21:
Ok, here's an interesting thing: I tried [this](https://github.com/liip/LiipImagineBundle/issues/784#issuecomment-287739886) which gave:

[image]

So the thumbnail generator works and generates thumbnails correctly. However, `https://partdb.mydomain.nl/media/cache/resolve/thumbnail_md/media/user/2/avatar-64e37182c18d8.jpeg` still results in a 404. If I remove the `resolve/` part I get the resized image(s).

RobThree, 2023-08-21:
Ok, I found it. It turned out that one of the includes (`include h5bp/basic_extrasec.conf;`) included nginx config for handling images etc. which I overlooked. I got rid of it and it started working immediately. It's entirely my bad. Sorry for wasting your time.

Having said that, I still get this error:

[image]

It seems this CSP is set in code? Or am I mistaken?

RobThree, 2023-08-21:
Damnit... After some digging it turns out it's [Read it later](https://chrome.google.com/webstore/detail/save-to-pocket/niloccemoadcdkdjlinkgdfekeahmflj); it only does that on mydomain.nl which is weird but none of your worries. I'll go figure out what going on there. So that's 2 for 2: both my bad, both my mistake. I am terribly sorry for wasting your time. My apologies.

closed 2023-08-21 by RobThree