Part-DB/Part-DB-server #354: Update failed, nonexistent parameter partdb.gpdr_compliance

virtadpt, 2023-08-22:
**Describe the bug**
When trying to update from 1.6.0-dev to commit 098fcb29fbb1be19e04bcf13241150bd2d77b1d4, I get the following error message when running `composer update --ignore-platform-req=ext-intl` early in the procedure:

```
Script cache:clear returned with error code 1
!!
!!  In DefinitionErrorExceptionPass.php line 49:
!!
!!    You have requested a non-existent parameter "partdb.gdpr_compliance". Did y
!!    ou mean this: "partdb.gpdr_compliance"?
!!
!!
!!
Script @auto-scripts was called via post-update-cmd
```

`vendor/` is created, `node_modules/` isn't because the update process doesn't get that far.

It looks like a typo in the code - `partdb.gpdr_compliance` instead of `partdb.gdpr_compliance`

**To Reproduce**
Steps to reproduce the behavior:
1. `cd` to local clone of Git repo.
2. `git pull`
3. `rm -rf vendor/ node_modules/`
4. `composer update --ignore-platform-req=ext-intl`
5. See error.

**Expected behavior**
I should see "Executing script cache:clear [KO]" (because I can't build Part-DB-server on Dreamhost's web servers) but everything else should work.

**Server Side**
- Part-DB Version: Commit 098fcb29fbb1be19e04bcf13241150bd2d77b1d4
- PHP Version: PHP 8.2.9 (cli)
- Database Server: mysql from 11.0.2-MariaDB

**Additional context**
I build on a home server and rsync the finished version up to Dreamhost's web server because they don't have composer, yarn or any dependencies of same installed.

jbtronics, 2023-08-22:
This is most likely an issue in your parameters.yaml (the Parameter was renamed a while ago, to fix the typo and the old parameters.yaml will not work)

Ensure that this is the version from the git repo and not some local changed version (you should be able to see that with `git status`)

Delete the `var/cache/` folder and try it again.

virtadpt, 2023-08-22:
I did a new `git clone` of this repo (mine was old enough that it was still called Part-DB-symfony), configured .env.local, left parameters.yaml alone, and went through the process again.  Different problem - (see ticket #316), solved by looking up how to specify the MySQL database server version in the DATABASE_URL variable in .env.local.

closed 2023-09-09 by jbtronics