Part-DB/Part-DB-server #362: Enable SSL in Docker

ewa, 2023-09-05:
So, I started looking at this because I was having problems with DigiKey (See issue #326, https://github.com/Part-DB/Part-DB-server/issues/326#issuecomment-1707264446).  It tuns out I still have DigiKey problems, but at least now I have SSL, too. 

If this is of interest, I can submit a proper merge request, but the relevant changes are: https://github.com/Part-DB/Part-DB-server/compare/master...ewa:Part-DB-server:master .  Primarily, there is an ssl version of `symfony.conf` and there is a directory mapping `./.docker/confs/ssl:/etc/apache2/ssl` which provides the container with (user-generated) ssl keys.  Naturally, such keys are unsigned or self-signed and will not be trusted by a browser until/unless the user configures their system or browser to do so, but that's a separate problem.

This makes `https://localhost/en/` a valid URL for the container, and means that you can have an `https` url for (e.g.) a DigiKey app OAuth redirect URL (i.e. `
https://localhost/en/oauth/client/ip_digikey_oauth/check`), though that doesn't necessarily make everything work with DigiKey.

jbtronics, 2023-09-05:
I wonder if it might be more comfortable, to provide a docker-compose with a reverse proxy frontend with traefik that can automatically configure letsencrypt certificates: https://doc.traefik.io/traefik/user-guides/docker-compose/acme-tls/

Also this would have the advantage that it makes the TLS encryption independent from apache. I want to release an (experimental) docker image which uses RoadrunnerPHP in the future. Roadrunner has the potential to make Part-DB much faster than on a classic web server with PHP-fpm, as it can eliminate the overhead of initializing the whole framework on each request.
But as roadrunner spawns it own webserver, there is no need to use apache and therefore You cannot use apaches ssl config (unless you want to use it as reverse proxy, but then you can use traefik too).

ewa, 2023-09-08:
Hmm, I don't have any knowledge about roadrunner or traefik, so I can't say much there.  For me, I would prefer for it not to depending on having a reverse-proxied, or otherwise globally-reachable, configuration.  In my use case, where I'm basically a one-person operation, being able to run in a local-only mode is a good thing.  And I _believe_ OAuth should be fine with that, though I haven't looked at it in quite a while.

That said, anything's fine with me!