Part-DB/Part-DB-server #363: Parameters & Paths are not imported correctly

stdlogicvector, 2023-09-05:
Importing of Parameters doesn't seem to work, neither from CSV nor from JSON.

In the CSV I tried setting the columns 'parameters.0.name' and 'parameters.0.value_text' but the are ignored.

In JSON I added the following array

```
"parameters": [
            {
                "name": "Param1",
                "value_typical": 12.34
            },
            {
                "name": "Param2",
                "value_text": "Value"
            }
        ]
```
but it is also ignored.

Additionally, paths don't seem to work when importing JSON files.

When I imported a part structure containing the following data

```
"category": {
            "type": "category",
            "name": "Fictional Books",
            "full_name": "Books->Fictional Books"
        },
        "footprint": {
            "type": "footprint",
            "name": "Hardcover",
            "full_name": "Book->Hardcover"
        },

```
I got new categories named "Books->Factual Books" (even though "Create unknown datastructures" was disabled), instead of adding the part to the existing subcategory "Fictional Books" of the topcategory "Books". The same problem with the footprints.

I'm using the 1.7.0 Release of PartDB.

jbtronics, 2023-09-05:
In general not all fields that get exported, can be imported again. The Parameters and attachments field are not marked as importable. I'm not sure if there was a problem with preventing it or if it was just forgotten. For attachments that's a bit more difficult, as attachments without files are a bit pointless.

That the full_name field of a data structure is not respected, is caused by the fact that full_name is just a virtual field, which is generated on export, and not a real value in the database entry. It works if you pass a string with the full path to the "category" (or similar) instead of a object. In a CSV file you would just put the full path as string in the category column.
I will check if it is possible to make export/import more idempotent, by allowing to use the full_path field for imports.

In general nested imports (passing a object as property of a part) are pretty difficult, especially if a similar object already exists in the database... So there might be no really good solution (at least not if the system should still be flexible and independent of database specific IDs)

stdlogicvector, 2023-09-06:
Thanks for the explanation!
The documentation sounded like importing of nested structures was possible with JSON.

The category import is fine the way it is! (It's just not clearly documented). And I concur on the pointlessness of importing attachments. Maybe with base64-encoded files in the JSON...? Just kidding!

But the parameter import is something I would really like.
If necessary only for parts that are newly created by the import.

It's so much easier to gather all the data in a spreadsheet and then import it instead of endlessly clicking through the forms.
I wouldn't mind if the complex import only supported JSON. Converting a spreadsheet to JSON isn't a problem.

jbtronics, 2023-09-09:
> But the parameter import is something I would really like. If necessary only for parts that are newly created by the import.
>
> It's so much easier to gather all the data in a spreadsheet and then [...]

I will look into that, when working on the API interfaces to interact with parameters, as the import/export system and the API both use the same system (Symfony Serializer) to convert JSON into database objects and vice versa.

jbtronics, 2024-06-22:
Parameter import is now possible.

closed 2024-06-22 by jbtronics

NikoDelarich, 2024-07-23:
Thanks for taking the time to implement this! However, I'm still unable to import parameters in version 1.13.1.
Here's what I'm doing:
- Create part with parameters
- Export part ("Full")
- Delete part
- Import part

After the import, the part is created but the parameters are missing. Do I need to change anything in the import file?
I already tried all the different import formats (JSON, CSV, XML, YAML)...

stdlogicvector, 2025-01-18:
In Version 1.15.1 Parameter Import works for me via CSV with columns named _parameters.0.name_ and _parameters.0.value_typical_