Part-DB/Part-DB-server #364: Showing subparts in the BOM (Sub-BOM)

cAbtHelectronics, 2023-09-06:
We got a lot of projects which use parts created from subprojects. Now when I want to build the project or show the BOM I will only see the parts of the main-project. It would be great if I could show the sub-BOM of the part from the subproject within the main-BOM. 

A solution would be the Drop-Down-Menu which can be clicked when I want to the sub-BOM of a part in the main-BOM. 

How Part-DB shows it now:
```
Project A:
- Part A (2x)
- Part B (1x) [subproject]
- Part C (10x)
```

How it could be:
```
Project A:
- Part A (2x)
- Part B (1x) [subproject]
   - Part D (4x)
   - Part E (1x)
- Part C (10x)
```

Additional would it be great to see if parts of subprojects are missing when I want to build the main-project.

kookon, 2024-11-14:
I would add that for me this is a crucial feature. When building the main project there should be a possibility to add sub-project parts as well to the build because they may contain (and usually do) same parts. 

For example when I would like to order parts for both projects, I have to compare builds because I might have parts for either but not for both.

Or the other possibility would be to build multiple projects at the same time (like it was in Partkeepr).

sbrown7792, 2025-07-08:
I want to add my support for this enhancement, as my use-case seems to align and overlap quite a bit.

If I have:
- enough BOM parts to build qty 5 of Project A (_but no "project builds part" of that in stock_),
- enough BOM parts to build qty 5 of Project B (_but no "project builds part" of that in stock_),
- Project AB whose BOM contains only qty 1 part of each Project A and Project B... 

I'd like the "Build" tab of Project AB to show the ability to build qty 5, with the added context that I'd have to first build the sub-project A and B first. And this would apply recursively, as I have some projects that are 5+ sub-projects deep with various sub-assemblies, and would like to know how many top-level parts of a project I _could_ build at any time, even if it would require sub-projects to be built up first.

Thanks!